meta-flutter/flutter_embedded #1: Build failed while trying to build libcxxabi

jakesays, 2018-12-19:
here's the output:

    "/home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/jake/pax/flutter_embedded/build/libcxxabi-prefix/src/libcxxabi-build/CMakeFiles/CMakeTmp

    Run Build Command:"/usr/bin/make" "cmTC_f318f/fast"
    /usr/bin/make -f CMakeFiles/cmTC_f318f.dir/build.make CMakeFiles/cmTC_f318f.dir/build
    make[1]: Entering directory '/home/jake/pax/flutter_embedded/build/libcxxabi-prefix/src/libcxxabi-build/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_f318f.dir/testCXXCompiler.cxx.o
    /home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang++ --sysroot=/home/jake/pax/flutter_embedded/sdk/sysroot    --target=arm-linux-gnueabihf -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS --target=arm-linux-gnueabihf -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS    -o CMakeFiles/cmTC_f318f.dir/testCXXCompiler.cxx.o -c /home/jake/pax/flutter_embedded/build/libcxxabi-prefix/src/libcxxabi-build/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_f318f
    /opt/cmake/bin/cmake -E cmake_link_script CMakeFiles/cmTC_f318f.dir/link.txt --verbose=1
    /home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang++ --sysroot=/home/jake/pax/flutter_embedded/sdk/sysroot   --target=arm-linux-gnueabihf -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS --target=arm-linux-gnueabihf -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS     CMakeFiles/cmTC_f318f.dir/testCXXCompiler.cxx.o  -o cmTC_f318f
    /usr/bin/arm-linux-gnueabihf-ld: cannot find crt1.o: No such file or directory
    /usr/bin/arm-linux-gnueabihf-ld: cannot find crti.o: No such file or directory
    /usr/bin/arm-linux-gnueabihf-ld: cannot find crtbegin.o: No such file or directory
    /usr/bin/arm-linux-gnueabihf-ld: cannot find -lstdc++
    /usr/bin/arm-linux-gnueabihf-ld: cannot find -lm
    /usr/bin/arm-linux-gnueabihf-ld: cannot find -lgcc_s
    /usr/bin/arm-linux-gnueabihf-ld: cannot find -lgcc
    /usr/bin/arm-linux-gnueabihf-ld: cannot find -lc
    /usr/bin/arm-linux-gnueabihf-ld: cannot find -lgcc_s
    /usr/bin/arm-linux-gnueabihf-ld: cannot find -lgcc
    /usr/bin/arm-linux-gnueabihf-ld: cannot find crtend.o: No such file or directory
    /usr/bin/arm-linux-gnueabihf-ld: cannot find crtn.o: No such file or directory
    clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [CMakeFiles/cmTC_f318f.dir/build.make:87: cmTC_f318f] Error 1
    make[1]: Leaving directory '/home/jake/pax/flutter_embedded/build/libcxxabi-prefix/src/libcxxabi-build/CMakeFiles/CMakeTmp'
    make: *** [Makefile:121: cmTC_f318f/fast] Error 2

jwinarske, 2018-12-22:
If you pull master now, it will work with a libcxxabi, and libcxx workaround applied.

Once I hear back from the libcxx-dev mailing list, I'll either generate a PR for libcxxabi/libcxx; or implement a patch.

If you pull master, and get blocked, I can provide the workaround.

jwinarske, 2018-12-23:
Actually, I think I found your issue.  I just built on a new machine without the sysroot present, and I get the same results as you.  The fix was a matter of setting up the sysroot folder.

### Raspberry Pi sysroot example
Download [Raspbian Lite image](https://downloads.raspberrypi.org/raspbian_lite_latest) from [Raspbian Image Download Page](https://www.raspberrypi.org/downloads/raspbian/).  Extract the image, and you will have a .img file.  If on Ubuntu Bionic, double click the img file.  It will auto-mount the partitions present.  You're interested in the rootfs partition.  From terminal or nautilus, copy the /lib, /opt, and /usr directories from the rootfs mount.  Copy these to the default sysroot folder - {flutter_embedded git clone root}/sdk/sysroot

It would look like this after clang and binutils built correctly:

    ./sdk
    |-- sysroot
    |   |-- lib
    |   |-- opt
    |   `-- usr
    `-- toolchain
        |-- arm-linux-gnueabihf
        |-- bin
        |-- include
        |-- lib
        |-- libexec
        `-- share

You can optionally override the default sysroot location using a CMake variable.

jakesays, 2018-12-28:
As a test I built with a sysroot and default settings and it worked. What I really need, though is arm-linux-gnueabi. I'm targeting a very old device (linux 3.0) with 64MB (yes that's an M. lol). Should be fun.

I'll try it again with the the proper target values set for armel.

jwinarske, 2019-01-03:
Sounds like fun :)

closed 2019-01-03 by jwinarske