meta-flutter/flutter_embedded #2: No such file or directory

pauldemarco, 2019-01-07:
When running this command:
`cmake .. -DCMAKE_BUILD_TYPE=Release -GNinja`

I get this:
```
-- The CXX compiler identification is GNU 7.3.0
-- The C compiler identification is GNU 7.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Generator .............. Ninja
-- Build Type ............. Release
-- llvm-config ............ /home/paul/flutter_embedded/sdk/toolchain/bin/llvm-config
-- /home/paul/flutter_embedded/sdk/toolchain/bin/llvm-config --cflags --cxxflags --ldflags
CMake Error at cmake/llvm_config.cmake:20 (message):
  llvm-config failed with status No such file or directory
Call Stack (most recent call first):
  cmake/build_dependencies.cmake:39 (include)
  CMakeLists.txt:48 (include)

-- Configuring incomplete, errors occurred!
See also "/home/paul/flutter_embedded/build/CMakeFiles/CMakeOutput.log".
```

pauldemarco, 2019-01-08:
The location of llvm-config can be specified with: `-DLLVM_CONFIG_PATH=path/to/llvm-config`

closed 2019-01-08 by pauldemarco

jwinarske, 2019-01-10:
Until I add the patch files for LLVM, this won't build out of the box.  The issue is with LLVM projects and CMAKE_TOOLCHAIN_FILE, which are claimed to be supported; there's an issue.  So until solution is implemented for LLVM trunk, you will need a patch.  I should have that checked in, in a couple of days.

jakesays, 2019-01-14:
@jwinarske I just ran in to this error. I'm not sure how this is an LLVM issue because it happens before the llvm source is downloaded.

jwinarske, 2019-01-14:
@jakesays llvm_config is used only after the dependency of clang is met.  It's okay to define the path variable before it's used.

The stack wasn't building error free until recently.  Be sure to issue a git pull, and start with a fresh build folder.  Start up another issue if you hit any issues, and be sure to let me know your environment.

jakesays, 2019-01-14:
@jwinarske that's not true - you include(llvm_config.cmake) way before the clang dep has been met. That's why it's failing.