meta-flutter/flutter_embedded #3: LLVM_CONFIG_PATH being used before llvm-config exists

jakesays, 2019-01-14:
build_dependencies.cmake is using LLVM_CONFIG_PATH before llvm-config is built.

```cmake
if(NOT ANDROID)

   ...

    if(NOT LLVM_CONFIG_PATH)
        set(LLVM_CONFIG_PATH ${TOOLCHAIN_DIR}/bin/llvm-config CACHE PATH "llvm-config path")
    endif()
    include(llvm_config)

   ...

endif()
```

This causes the build to fail.

jwinarske, 2019-01-14:
All the projects that use the variable LLVM_CONFIG_PATH, have a dependency on "clang" and "binutils".  Meaning this variable isn't used by anything until Clang has built.

What are you using for your cmake cmd line?

jakesays, 2019-01-14:
cmake -DTARGET_TRIPLE=arm-arm1176jzs-linux-gnueabi -GNinja -DCMAKE_BUILD_TYPE=Release -DLLVM_TARGETS_TO_BUILD=ARM -DTARGET_ARCH=armv6kz -DBUILD_TSLIB=OFF -DENGINE_RUNTIME_MODE=debug

The variable is used by llvm_config.cmake  - you `include(llvm_config)`, and llvm_config.cmake immediately tries to execute llvm-config to set up variables.

I understand what you're saying about dependencies, but that isn't correct - because llvm_config.cmake *is* dependent on LLVM_CONFIG_PATH.

jwinarske, 2019-01-15:
I just checked in a fix.  It's all around the app toolchain file.

If llvm_config_path is passed in, it will create app toolchain file right then, otherwise if toolchain is built, it will create app toolchain file clang post_build.