meta-flutter/flutter_embedded #4: Build failed due to missing header

jakesays, 2019-01-15:
The compiler-rt build step failed because it couldn't find `<cstdint>`. Below is the build output.

/home/jake/pax/depot_tools/ninja
[0/1] Re-running CMake...
-- Generator .............. Ninja
-- Build Type ............. Release
-- llvm-config ............ /home/jake/pax/flutter_embedded/sdk/toolchain/bin/llvm-config
-- Flutter ................
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jake/pax/flutter_embedded/out
[1/36] Performing configure step for 'compiler-rt'
-- The C compiler identification is Clang 8.0.0
-- The CXX compiler identification is Clang 8.0.0
-- The ASM compiler identification is Clang
-- Found assembler: /home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang
-- Check for working C compiler: /home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang
-- Check for working C compiler: /home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang++
-- Check for working CXX compiler: /home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for unwind.h
-- Looking for unwind.h - found
-- Looking for rpc/xdr.h
-- Looking for rpc/xdr.h - found
CMake Warning at cmake/Modules/CompilerRTUtils.cmake:261 (message):
  testingsupport library not installed, some tests will be skipped
Call Stack (most recent call first):
  CMakeLists.txt:69 (load_llvm_config)

-- Found PythonInterp: /usr/bin/python2.7 (found version "2.7.15")
-- Looking for fopen in c
-- Looking for fopen in c - found
-- Looking for __gcc_personality_v0 in gcc_s
-- Looking for __gcc_personality_v0 in gcc_s - found
-- Performing Test COMPILER_RT_HAS_NODEFAULTLIBS_FLAG
-- Performing Test COMPILER_RT_HAS_NODEFAULTLIBS_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FFREESTANDING_FLAG
-- Performing Test COMPILER_RT_HAS_FFREESTANDING_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FPIE_FLAG
-- Performing Test COMPILER_RT_HAS_FPIE_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FNO_BUILTIN_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_BUILTIN_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FNO_EXCEPTIONS_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_EXCEPTIONS_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FOMIT_FRAME_POINTER_FLAG
-- Performing Test COMPILER_RT_HAS_FOMIT_FRAME_POINTER_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FUNWIND_TABLES_FLAG
-- Performing Test COMPILER_RT_HAS_FUNWIND_TABLES_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FNO_STACK_PROTECTOR_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_STACK_PROTECTOR_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FNO_SANITIZE_SAFE_STACK_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_SANITIZE_SAFE_STACK_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FVISIBILITY_HIDDEN_FLAG
-- Performing Test COMPILER_RT_HAS_FVISIBILITY_HIDDEN_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FRTTI_FLAG
-- Performing Test COMPILER_RT_HAS_FRTTI_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FNO_RTTI_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_RTTI_FLAG - Success
-- Performing Test COMPILER_RT_HAS_FNO_FUNCTION_SECTIONS_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_FUNCTION_SECTIONS_FLAG - Success
-- Performing Test COMPILER_RT_HAS_STD_CXX11_FLAG
-- Performing Test COMPILER_RT_HAS_STD_CXX11_FLAG - Success
[... 99 lines cut ...]
-- Looking for __VFP_FP__ - not found
-- Performing Test COMPILER_RT_TARGET_HAS_ATOMICS
-- Performing Test COMPILER_RT_TARGET_HAS_ATOMICS - Success
-- Performing Test COMPILER_RT_TARGET_HAS_FCNTL_LCK
-- Performing Test COMPILER_RT_TARGET_HAS_FCNTL_LCK - Success
-- Performing Test COMPILER_RT_TARGET_HAS_UNAME
-- Performing Test COMPILER_RT_TARGET_HAS_UNAME - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jake/pax/flutter_embedded/out/compiler-rt-prefix/src/compiler-rt-build
[2/36] Performing build step for 'compiler-rt'
FAILED: compiler-rt-prefix/src/compiler-rt-stamp/compiler-rt-build
cd /home/jake/pax/flutter_embedded/out/compiler-rt-prefix/src/compiler-rt-build && /opt/cmake/bin/cmake --build . && /opt/cmake/bin/cmake -E touch /home/jake/pax/flutter_embedded/out/compiler-rt-prefix/src/compiler-rt-stamp/compiler-rt-build
Scanning dependencies of target RTXrayBASIC.arm
[  0%] Building CXX object lib/xray/CMakeFiles/RTXrayBASIC.arm.dir/xray_basic_flags.cc.o
[  0%] Building CXX object lib/xray/CMakeFiles/RTXrayBASIC.arm.dir/xray_basic_logging.cc.o
In file included from /home/jake/pax/flutter_embedded/out/llvm/projects/compiler-rt/lib/xray/xray_basic_logging.cc:31:
/home/jake/pax/flutter_embedded/out/llvm/projects/compiler-rt/lib/xray/../../include/xray/xray_records.h:20:10: fatal error: 'cstdint' file not found
#include <cstdint>
         ^~~~~~~~~
1 error generated.
make[2]: *** [lib/xray/CMakeFiles/RTXrayBASIC.arm.dir/build.make:76: lib/xray/CMakeFiles/RTXrayBASIC.arm.dir/xray_basic_logging.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:5365: lib/xray/CMakeFiles/RTXrayBASIC.arm.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
ninja: build stopped: subcommand failed.

jwinarske, 2019-01-15:
I'm thinking a missing include path.

In build_dependencies.cmake in the compiler-rt project, change "-DCMAKE_VERBOSE_MAKEFILE=${CMAKE_VERBOSE_MAKEFILE}" to "-DCMAKE_VERBOSE_MAKEFILE=TRUE".  This will enable verbose without rebuilding clang.  Provide this output for compiler-rt.

Provide output for /home/jake/pax/flutter_embedded/sdk/toolchain/bin/llvm-config --cxflags

Additions to include path, to affect compiler-rt, you would add it to CMAKE_CXX_FLAGS, in cmake/clang.toolchain.cmake.in

If you have the same sysroot somewhere I can download, I can give it a shot if you like.

jakesays, 2019-01-15:
I know where the headers are in my sysroot, but since I'm building libcxx, isn't compiler-rt supposed to be using the libcxx headers?

jwinarske, 2019-01-15:
Perhaps this will help.  This is the verbose output of building for Raspberry Pi, for the first file in compiler-rt:

```
[  0%] Building CXX object lib/xray/CMakeFiles/RTXrayBASIC.armhf.dir/xray_basic_flags.cc.o
cd /home/joel/git/flutter_embedded/build/compiler-rt-prefix/src/compiler-rt-build/lib/xray && /home/joel/git/flutter_embedded/sdk/toolchain/bin/clang++ --sysroot=/home/joel/git/flutter_embedded/sdk/sysroot  -DXRAY_HAS_EXCEPTIONS=1 -DXRAY_SUPPORTED=1 -I/home/joel/git/flutter_embedded/build/llvm/projects/
compiler-rt/lib/xray/.. -I/home/joel/git/flutter_embedded/build/llvm/projects/compiler-rt/lib/xray/../../include  --target=arm-linux-gnueabihf -v --target=arm-linux-gnueabihf -v -Wall -std=c++11 -Wno-unused-parameter -Os -DNDEBUG    -march=armv7-a -mfloat-abi=hard -fPIC -fno-builtin -fno-exceptions -fom
it-frame-pointer -fno-stack-protector -fno-sanitize=safe-stack -fvisibility=hidden -fno-lto -O3 -gline-tables-only -Wno-gnu -Wno-variadic-macros -Wno-c99-extensions -Wno-non-virtual-dtor -fno-rtti -o CMakeFiles/RTXrayBASIC.armhf.dir/xray_basic_flags.cc.o -c /home/joel/git/flutter_embedded/build/llvm/pro
jects/compiler-rt/lib/xray/xray_basic_flags.cc

clang version 8.0.0 (trunk 351033)
Target: arm-unknown-linux-gnueabihf
Thread model: posix
InstalledDir: /home/joel/git/flutter_embedded/sdk/toolchain/bin
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/4.6
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/4.6.4
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/4.7
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/4.7.3
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/4.8
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/4.8.5
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/4.9
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/4.9.3
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/5
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/5.4.1
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/6
Found candidate GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/6.3.0
Selected GCC installation: /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/6.3.0
Candidate multilib: .;@m32
Selected multilib: .;@m32
 "/home/joel/git/flutter_embedded/sdk/toolchain/bin/clang-8" -cc1 -triple armv7-unknown-linux-gnueabihf -emit-obj -disable-free -disable-llvm-verifier -discard-value-names -main-file-name xray_basic_flags.cc -mrelocation-model pic -pic-level 2 -mthread-model posix -fmath-errno -masm-verbose -mconstructor-aliases -fuse-init-array -target-cpu generic -target-abi aapcs-linux -mfloat-abi hard -fallow-half-arguments-and-returns -dwarf-column-info -debug-info-kind=line-tables-only -dwarf-version=4 -debugger-tuning=gdb -v -v -coverage-notes-file /home/joel/git/flutter_embedded/build/compiler-rt-prefix/src/compiler-rt-build/lib/xray/CMakeFiles/RTXrayBASIC.armhf.dir/xray_basic_flags.cc.gcno -resource-dir /home/joel/git/flutter_embedded/sdk/toolchain/lib/clang/8.0.0 -D XRAY_HAS_EXCEPTIONS=1 -D XRAY_SUPPORTED=1 -I /home/joel/git/flutter_embedded/build/llvm/projects/compiler-rt/lib/xray/.. -I /home/joel/git/flutter_embedded/build/llvm/projects/compiler-rt/lib/xray/../../include -D NDEBUG -isysroot /home/joel/git/flutter_embedded/sdk/sysroot -internal-isystem /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/6.3.0/../../../../include/c++/6.3.0 -internal-isystem /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/6.3.0/../../../../include/arm-linux-gnueabihf/c++/6.3.0 -internal-isystem /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/6.3.0/../../../../include/arm-linux-gnueabihf/c++/6.3.0 -internal-isystem /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/6.3.0/../../../../include/c++/6.3.0/backward -internal-isystem /home/joel/git/flutter_embedded/sdk/sysroot/usr/local/include -internal-isystem /home/joel/git/flutter_embedded/sdk/toolchain/lib/clang/8.0.0/include -internal-externc-isystem /home/joel/git/flutter_embedded/sdk/sysroot/usr/include/arm-linux-gnueabihf -internal-externc-isystem /home/joel/git/flutter_embedded/sdk/sysroot/include -internal-externc-isystem /home/joel/git/flutter_embedded/sdk/sysroot/usr/include -O3 -Wall -Wno-unused-parameter -Wno-gnu -Wno-variadic-macros -Wno-c99-extensions -Wno-non-virtual-dtor -std=c++11 -fdeprecated-macro -fdebug-compilation-dir /home/joel/git/flutter_embedded/build/compiler-rt-prefix/src/compiler-rt-build/lib/xray -ferror-limit 19 -fmessage-length 304 -fvisibility hidden -fno-builtin -fno-rtti -fno-signed-char -fobjc-runtime=gcc -fdiagnostics-show-option -fcolor-diagnostics -vectorize-loops -vectorize-slp -o CMakeFiles/RTXrayBASIC.armhf.dir/xray_basic_flags.cc.o -x c++ /home/joel/git/flutter_embedded/build/llvm/projects/compiler-rt/lib/xray/xray_bas [... cut ...]
 /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/6.3.0/../../../../include/arm-linux-gnueabihf/c++/6.3.0
 /home/joel/git/flutter_embedded/sdk/sysroot/usr/lib/gcc/arm-linux-gnueabihf/6.3.0/../../../../include/c++/6.3.0/backward
 /home/joel/git/flutter_embedded/sdk/sysroot/usr/local/include
 /home/joel/git/flutter_embedded/sdk/toolchain/lib/clang/8.0.0/include
 /home/joel/git/flutter_embedded/sdk/sysroot/usr/include/arm-linux-gnueabihf
 /home/joel/git/flutter_embedded/sdk/sysroot/include
 /home/joel/git/flutter_embedded/sdk/sysroot/usr/include
End of search list.

```

jakesays, 2019-01-15:
Ohh libc++ headers are under sdk/toolchain. Makes sense.

Only difference is mine are under `sys/toolchain/include/c++/v1/cstdlib`.

jakesays, 2019-01-15:
It' sounds like we are working on very similar projects. I'd like to compare notes. Do you have an email address?

jwinarske, 2019-01-15:
The toolchain is built with gcc.  The app build uses the flag -stdlib=libc++, which then selects the proper headers, and lib.

jwinarske, 2019-01-15:
@jakesays joel.winarske@gmail.com

jakesays, 2019-01-15:
@jwinarske ah. i thought the support libs were supposed to be built with the previously built clang.

jakesays, 2019-01-15:
It looks like the new toolchain is being used to build the support libs. wouldn't it make sense to use libc++ headers instead of the g++ headers?

jwinarske, 2019-01-15:
@jakesays I just pushed.  Try that.

jakesays, 2019-01-15:
I pulled and built. Here's my error log (for the record I don't like cmake. it's basically configure+make with different syntax. I much prefer gn, where I can tell it exactly which tools to use and their settings, etc).

Two thinks of note:
1) the first two failures are using the wrong linker
2) the last failure i'm pretty sure is due to not linking against the new compiler_rt

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang 
Build flags: ;--target=arm-arm1176jzs-linux-gnueabi;--target=arm-arm1176jzs-linux-gnueabi
Id flags:  

The output was:
1
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find crt1.o: No such file or directory
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find crti.o: No such file or directory
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find crtbegin.o: No such file or directory
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lgcc
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lgcc_s
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lc
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lgcc
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lgcc_s
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find crtend.o: No such file or directory
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find crtn.o: No such file or directory
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang++ 
Build flags: ;--target=arm-arm1176jzs-linux-gnueabi;-I/home/jake/pax/flutter_embedded/sdk/sysroot/usr/include/c++/8.2.0;-I/home/jake/pax/flutter_embedded/sdk/sysroot/usr/include/c++/8.2.0/arm-arm1176jzs-linux-gnueabi;--target=arm-arm1176jzs-linux-gnueabi;-I/home/jake/pax/flutter_embedded/sdk/sysroot/usr/include/c++/8.2.0;-I/home/jake/pax/flutter_embedded/sdk/sysroot/usr/include/c++/8.2.0/arm-arm1176jzs-linux-gnueabi
Id flags:  

The output was:
1
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find crt1.o: No such file or directory
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find crti.o: No such file or directory
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find crtbegin.o: No such file or directory
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lstdc++
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lm
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lgcc_s
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lgcc
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lc
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lgcc_s
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find -lgcc
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find crtend.o: No such file or directory
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/arm-arm1176jzs-linux-gnueabi-ld: cannot find crtn.o: No such file or directory
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)

Performing C++ SOURCE FILE Test LLVM_NO_OLD_LIBSTDCXX failed with the following output:
Change Dir: /home/jake/pax/flutter_embedded/out/libunwind-prefix/src/libunwind-build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_fe24a/fast"
/usr/bin/make -f CMakeFiles/cmTC_fe24a.dir/build.make CMakeFiles/cmTC_fe24a.dir/build
make[1]: Entering directory '/home/jake/pax/flutter_embedded/out/libunwind-prefix/src/libunwind-build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_fe24a.dir/src.cxx.o
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang++ --sysroot=/home/jake/pax/flutter_embedded/sdk/sysroot    --target=arm-arm1176jzs-linux-gnueabi -I/home/jake/pax/flutter_embedded/sdk/sysroot/usr/include/c++/8.2.0 -I/home/jake/pax/flutter_embedded/sdk/sysroot/usr/include/c++/8.2.0/arm-arm1176jzs-linux-gnueabi --target=arm-arm1176jzs-linux-gnueabi -I/home/jake/pax/flutter_embedded/sdk/sysroot/usr/include/c++/8.2.0 -I/home/jake/pax/flutter_embedded/sdk/sysroot/usr/include/c++/8.2.0/arm-arm1176jzs-linux-gnueabi -DLLVM_NO_OLD_LIBSTDCXX  -std=c++0x   -o CMakeFiles/cmTC_fe24a.dir/src.cxx.o -c /home/jake/pax/flutter_embedded/out/libunwind-prefix/src/libunwind-build/CMakeFiles/CMakeTmp/src.cxx
Linking CXX executable cmTC_fe24a
/opt/cmake/bin/cmake -E cmake_link_script CMakeFiles/cmTC_fe24a.dir/link.txt --verbose=1
/home/jake/pax/flutter_embedded/sdk/toolchain/bin/clang++ --sysroot=/home/jake/pax/flutter_embedded/sdk/sysroot   --target=arm-arm1176jzs-linux-gnueabi -I/home/jake/pax/flutter_embedded/sdk/sysroot/usr/include/c++/8.2.0 -I [... cut ...]
CMakeFiles/cmTC_fe24a.dir/src.cxx.o:src.cxx:function __clang_call_terminate: error: undefined reference to 'std::terminate()'
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [CMakeFiles/cmTC_fe24a.dir/build.make:87: cmTC_fe24a] Error 1
make[1]: Leaving directory '/home/jake/pax/flutter_embedded/out/libunwind-prefix/src/libunwind-build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_fe24a/fast] Error 2

Source file was:

#include <atomic>
std::atomic<float> x(0.0f);
int main() { return (float)x; }

jwinarske, 2019-01-15:
@jakesays Well the last change I made only applies if this is *not* a bootstrap build, as the headers don't exist until the Install step of libcxx.  I'm starting clean build folder tests now, will implement travis-ci soon.  If you have a public sysroot, I can add that as a test case...

jakesays, 2019-01-16:
Sounds good. I've decided to take a brief detour and experiment with bionic instead of libc. 

My sysroot is an image of an old raspberry pi 1 soft float raspbian. the pi 1 minus floating point and a gpu is the same model as my target device.

jwinarske, 2019-03-27:
Not planning to support unique cases for now.  I will consider full support for cases where there is something publicly available.  I am available for consult work.

closed 2019-03-27 by jwinarske

jakesays, 2019-03-27:
I'm curious - how is this a unique case?

jwinarske, 2019-03-28:
Open a new case requesting support with the hardware software combo.  If there is an off the shelf device it would run on, be sure to include that.

Next priority for me will be adding framebuffer support in the next couple of weeks.