meta-flutter/flutter_embedded #5: Break up in to separate parts

jakesays, 2019-01-17:
Would it be possible to split this in to separate parts?

Suggestions:

- build toolchain
- build flutter engine
- build rpi host
- build pi userland

As it currently stands I cannot build just the toolchain without also building flutter. It would be great to have cmake files for just the toolchain without dependencies on anything else.

jwinarske, 2019-01-17:
Yeah.  At some point I'm going to factor LLVM related items out into a separate repo, with added config knobs for a bare metal needs, etc.

I started some re-factoring (see the cleanup branch), then got side tracked with sysroot automation, and toolchain variable caching.

What I have right now in master is ready for Appveyor and Travis-CI work.  I'll be looking at the Travis yaml on Monday.

jwinarske, 2019-01-29:
https://github.com/jwinarske/clang_toolchain

closed 2019-03-27 by jwinarske