meta-flutter/flutter_embedded #7: Support targeting a specific llvm branch

jakesays, 2019-01-17:
To overcome the recent llvm version bump the tools build should be able to target a specific llvm/clang/libs  branch.

jwinarske, 2019-01-18:
The problem that needs solving here, is setting the compiler-rt CMAKE_INSTALL_PREFIX to the proper Clang version.  

I experimented with doing this in the toolchain file, only setting CMAKE_INSTALL_PREFIX is ignored in the toolchain file.  It needs to get passed in.

My conclusion earlier today, is that this should really be handled in the compiler-rt project.  It has the version info available (I think).  I'm adding this to my LLVM upstream agenda.

I'm open to an interim workaround, if you have something in mind.

jakesays, 2019-01-18:
I just added to my PR support for a specific llvm branch, which should handle the install prefix issue.

If you set `-DUSE_LLVM_BRANCH=release_80` it will check out that branch. it defaults to trunk.

closed 2019-03-27 by jwinarske