meta-flutter/flutter_embedded #9: Raspberry pi

gedw99, 2019-03-22:
This is great.

The plugins work has already been done and works on desktops. It calls out to golang but c/c++ should be possible.
https://github.com/go-flutter-desktop/go-flutter
It's all been developed for golang.
Golang runs on all the embedded devices you are using also btw !

I really want to try out raspberry pi. How was the performance with larger apps ?

jwinarske, 2019-03-23:
Yeah I'm aware of the desktop project.  My goal is limited RAM devices.

Performance isn't bad.  I think it would work well for quite a few projects.

gedw99, 2019-03-25:
I also used to do QT, and i totally get you about the licensing.
In case you are  curious, you can compile golang to embedded using tinygo.
https://github.com/tinygo-org/tinygo

The list of supported boards is pretty good. Of course raspberryPi does not need this as golang natively supports raspberrypi.

Curious what you think.

jwinarske, 2019-03-27:
Yeah I've played around with Go a bit.  When I have some time (~3 weeks), I'll take a look at the Flutter desktop implementation, and see what the gaps are for SQLite, Bluetooth, GPS, Network adapter status, etc.

gedw99, 2019-05-21:
SQLite runs on go-flutter and Mobile flutter today.
https://github.com/go-flutter-desktop/plugins/blob/master/list.md

i actually think using the go-flutter ( https://github.com/go-flutter-desktop  ) desktop is better. All the functionality that the main flutter desktop team expose is easy to expose in the golang version. Its just simple CGO.

Whats holding back go-flutter desktop is the flutter-desktop project actually. lutter-desktop have not exposed all the API's you see. Like streaming Plugin channels.
Al other Plugins work on go-flutter desktop.

The embedder CGO binding: https://github.com/go-flutter-desktop/go-flutter/tree/master/embedder

The embedder downloader: https://github.com/go-flutter-desktop/hover/blob/master/internal/enginecache/cache.go
- It would download the RaspberryPi embedder.

So all thats needed is to compile the Flutter engine for ARM v7 and that gets you there.

For the embedded OS packaging and provisioning i am using this: https://github.com/gokrazy
- Not everyones cup of tea