meta-flutter/flutter_embedded #11: Can't clone shallow from sourceware.org/git/binutils-gdb.git

drewcovi, 2019-05-30:
Running into an issue with git 2.21.0

Any workaround recommendations?

When attempting a `git clone --depth 1` i can reproduce this same issue...

```
[18/64] Performing download step (git clone) for 'binutils'
FAILED: binutils-prefix/src/binutils-stamp/binutils-download 
cd /Users/h101426/Development/flutter_embedded/third_party && /usr/local/Cellar/cmake/3.14.4/bin/cmake -P /Users/h101426/Development/flutter_embedded/build/toolchain-prefix/src/toolchain-build/binutils-prefix/tmp/binutils-gitclone.cmake && /usr/local/Cellar/cmake/3.14.4/bin/cmake -E touch /Users/h101426/Development/flutter_embedded/build/toolchain-prefix/src/toolchain-build/binutils-prefix/src/binutils-stamp/binutils-download
Cloning into 'binutils'...
fatal: dumb http transport does not support shallow capabilities
Cloning into 'binutils'...
fatal: dumb http transport does not support shallow capabilities
Cloning into 'binutils'...
fatal: dumb http transport does not support shallow capabilities
```

jwinarske, 2019-05-30:
You can change this line in submodule:

https://github.com/jwinarske/clang_toolchain/blob/master/CMakeLists.txt

Starting @ Line 255:
    ExternalProject_Add(binutils
        GIT_REPOSITORY git://sourceware.org/git/binutils-gdb.git
        GIT_TAG ${BINUTILS_TAG}
        GIT_SHALLOW ON <- change to OFF

drewcovi, 2019-05-30:
Is this an issue with sourceware?

jwinarske, 2019-05-30:
No idea.  It works fine on git version 2.17.1.

drewcovi, 2019-06-03:
closing for now, as I'm making more progress running on Ubuntu Bionic.

closed 2019-06-03 by drewcovi