meta-flutter/flutter_embedded #13: Unable to build in Ubuntu 18.04

drewcovi, 2019-06-03:
Appears my version of arm-linux-gnueabihf-ld doesn't support `--color-diagnostics`

```
[1/11] Building C object src/CMakeFiles/unwind_objects.dir/UnwindRegistersSave.S.o
[2/11] Building C object src/CMakeFiles/unwind_objects.dir/UnwindRegistersRestore.S.o
[3/11] Building C object src/CMakeFiles/unwind_objects.dir/Unwind-sjlj.c.o
[4/11] Building C object src/CMakeFiles/unwind_objects.dir/UnwindLevel1.c.o
[5/11] Building CXX object src/CMakeFiles/unwind_objects.dir/Unwind-seh.cpp.o
[6/11] Building C object src/CMakeFiles/unwind_objects.dir/UnwindLevel1-gcc-ext.c.o
[7/11] Building CXX object src/CMakeFiles/unwind_objects.dir/Unwind-EHABI.cpp.o
[8/11] Building CXX object src/CMakeFiles/unwind_objects.dir/libunwind.cpp.o
[9/11] Linking CXX static library lib/libunwind.a
[10/11] Linking CXX shared library lib/libunwind.so.1.0
FAILED: lib/libunwind.so.1.0
: && /home/drew/Development/flutter_embedded/sdk/toolchain/bin/clang++ --target=arm-linux-gnueabihf --sysroot=/home/drew/Development/flutter_embedded/sdk/sysroot -fPIC -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -std=c++11 -w -fdiagnostics-color -ffunction-sections -fdata-sections --target=arm-linux-gnueabihf --sysroot=/home/drew/Development/flutter_embedded/sdk/sysroot -std=c++11 -Os -DNDEBUG  -Wl,-z,defs -Wl,-z,nodelete -Wl,--color-diagnostics  --target=arm-linux-gnueabihf --sysroot=/home/drew/Development/flutter_embedded/sdk/sysroot -nodefaultlibs -shared -Wl,-soname,libunwind.so.1 -o lib/libunwind.so.1.0 src/CMakeFiles/unwind_objects.dir/libunwind.cpp.o src/CMakeFiles/unwind_objects.dir/Unwind-EHABI.cpp.o src/CMakeFiles/unwind_objects.dir/Unwind-seh.cpp.o src/CMakeFiles/unwind_objects.dir/UnwindLevel1.c.o src/CMakeFiles/unwind_objects.dir/UnwindLevel1-gcc-ext.c.o src/CMakeFiles/unwind_objects.dir/Unwind-sjlj.c.o src/CMakeFiles/unwind_objects.dir/UnwindRegistersRestore.S.o src/CMakeFiles/unwind_objects.dir/UnwindRegistersSave.S.o  -Wl,-rpath,"\$ORIGIN/../lib" -lc -lgcc_s -lgcc -ldl -lpthread && :
/home/drew/Development/flutter_embedded/sdk/toolchain/bin/arm-linux-gnueabihf-ld: unrecognized option '--color-diagnostics'
```

DisDis, 2019-06-03:
yes, it is very simple problem:) .
`grep -rnw './' -e '--color-diagnostics'`
you can find all files and fix it.
I build flutter(1.6 master) on ARMv7, but not run app ...debug, debug...

DisDis, 2019-06-03:
and next error fix
`sudo apt install texinfo`

drewcovi, 2019-06-03:
@DisDis aren't these ninja files generated? Ideally theres a way to fix at the source rather than the outputs...

DisDis, 2019-06-03:
@drewcovi 1 hour ago, I built on Ubuntu 18.04. The main problem is application crash. If I fix this, I fix the ninja.

DisDis, 2019-06-03:
but I can help you to built ;)

DisDis, 2019-06-03:
@drewcovi   Fix all 'not found -lGL, ....': change my nsysroot
```
# ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libGL.so.1 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libGL.so
# ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libgtk-3.so.0 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libgtk-3.so
# ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libgdk-3.so.0 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libgdk-3.so
# ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libatk-1.0.so.0 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libatk-1.0.so
# ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libgdk_pixbuf-2.0.so.0 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libgdk_pixbuf-2.0.so

#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libpangocairo-1.0.so.0 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libpangocairo-1.0.so
#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libpango-1.0.so.0 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libpango-1.0.so
#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libgio-2.0.so.0 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libgio-2.0.so
#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libgobject-2.0.so.0 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libgobject-2.0.so
#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/lib/arm-linux-gnueabihf/libglib-2.0.so.0 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/lib/arm-linux-gnueabihf/libglib-2.0.so
#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libX11.so.6 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libX11.so
#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libXcursor.so.1 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libXcursor.so
#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libXinerama.so.1 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libXinerama.so
#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libXrandr.so.2 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libXrandr.so
#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libXxf86vm.so.1 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libXxf86vm.so
#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/lib/arm-linux-gnueabihf/libpcre.so.3 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/lib/arm-linux-gnueabihf/libpcre.so
#ln -s /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/lib/arm-linux-gnueabihf/libpcre.so.3 /home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot/usr/lib/arm-linux-gnueabihf/libpcre.so
```

DisDis, 2019-06-03:
i use RasP 3A+ and 2019-04-08-raspbian-stretch.img

drewcovi, 2019-06-03:
@disdis do you need that image prior to building?

DisDis, 2019-06-03:
@drewcovi yes, I have installed many dev packages. i copy all files(lib,opt,usr) from my RasPi

drewcovi, 2019-06-03:
@disdis so this is where the README.md left me confused... 

At which step are we supposed to copy the files from the RasPi?

Steps I'm following below:
```
git clone https://github.com/jwinarske/flutter_embedded
cd flutter_embedded
mkdir build && cd build
cmake .. -DCMAKE_BUILD_TYPE=Release -GNinja
autoninja
```

DisDis, 2019-06-03:
By default, the version from 2018 is downloaded.

DisDis, 2019-06-03:
i exec
`cmake .. -DTARGET_SYSROOT=/home/dis/workspace/flutter/flutter_embedded/sdk/nsysroot -DENGINE_RUNTIME_MODE=debug -DCMAKE_BUILD_TYPE=Debug -GNinja`

drewcovi, 2019-06-03:
@DisDis so if you're okay with the 2018 image you can just run those commands, and fixing the color-diagnostics portion....

But if you want the latest you need to copy the sysroot files over first....

DisDis, 2019-06-03:
At first I tried with the image of 2018, but for some reason it did not work.

drewcovi, 2019-06-04:
@DisDis circling back today, going to run through the `autoninja` until it breaks and then remove the `--color-diagnostics` flag from the built resources... also using a custom sysroot with the latest... but had to boot it and install x11... README.md doesn't say you have to actually boot it up, but I assume that's an oversight.

DisDis, 2019-06-04:
@drewcovi I can send flutter armv7 (bin / lib). I'm very interested in running something on raspi.

DisDis, 2019-06-04:
@jwinarske How to change the engine version on fff80a0d95e92da3bca4b73b935de59337b620c9 (9 Apr 2019)?

DisDis, 2019-06-05:
@jwinarske I started flutter engine(lates version), but still have a crash.
flutter: ══╡ EXCEPTION CAUGHT BY WIDGETS LIBRARY ╞═══════════════════════════════════════════════════════════
flutter: The following NoSuchMethodError was thrown attaching to the render tree:
flutter: The method 'merge' was called on null.
flutter: Receiver: null
flutter: Tried calling: merge(null)
flutter: 
flutter: When the exception was thrown, this was the stack:
flutter: #0      Object.noSuchMethod (dart:core-patch/object_patch.dart:50:5)
flutter: #1      new ThemeData (package:flutter/src/material/theme_data.dart:225:34)
<cut>

The situation is such that you need to “replace” rpi_flutter with https://github.com/flutter/engine/blob/master/shell/platform/glfw/flutter_glfw.cc
I will try to bring to the working prototype and send PR.

DisDis, 2019-06-06:
@drewcovi   Flutter armv7 (bin,lib, run.sh) https://drive.google.com/file/d/1nJZfsF7ik08-0byeKFli6bplmDvDThD-/view?usp=sharing

drewcovi, 2019-06-06:
@DisDis  you are about to have a lot more friends

DisDis, 2019-06-06:
@drewcovi very low performance, about 5-6 frames per second :(
(examples/flutter_gallery)

drewcovi, 2019-06-06:
unfortunately I'm hitting this:

`/home/pi/flutter/bin/flutter: error while loading shared libraries: libc++abi.so.1: cannot open shared object file: No such file or directory`

DisDis, 2019-06-06:
@drewcovi sudo apt-get install libc++abi1

DisDis, 2019-06-06:
@jwinarske fix env{PATH} https://github.com/jwinarske/clang_toolchain/pull/3

DisDis, 2019-06-06:
@jwinarske How apply patch to 'build/toolchain-prefix/src/toolchain-build/libunwind-prefix/src/libunwind-build/CMakeCache.txt'?
replace:
`LINKER_SUPPORTS_COLOR_DIAGNOSTICS:INTERNAL=1`
to: 
`LINKER_SUPPORTS_COLOR_DIAGNOSTICS:INTERNAL=0`

drewcovi, 2019-06-06:
so close now:

```
flutter: ══╡ EXCEPTION CAUGHT BY WIDGETS LIBRARY ╞═══════════════════════════════════════════════════════════
flutter: The following assertion was thrown attaching to the render tree:
flutter: Unknown platform.
flutter: linux was not recognized as a target platform. Consider updating the list of TargetPlatforms to
flutter: include this platform.

```

DisDis, 2019-06-06:
@drewcovi https://github.com/jwinarske/flutter_embedded/issues/14#issuecomment-499635609

drewcovi, 2019-06-06:
@DisDis is that just another ENV variable? or where does that override go?

DisDis, 2019-06-06:
@drewcovi example:
flutter/examples/flutter_gallery/lib/main.dart
`
import 'package:flutter/material.dart';
import 'package:flutter/foundation.dart';

import 'gallery/app.dart';

void main() {
  debugDefaultTargetPlatformOverride = TargetPlatform.fuchsia;
  runApp(const GalleryApp());
}
`

drewcovi, 2019-06-07:
Seems like we're making headway... 
```
sudo DISPLAY=:0 TSLIB_CONFFILE=/etc/ts.conf TSLIB_PLUGINDIR=/home/pi/lib/ts TSLIB_CALIBFILE=/etc/pointercal LD_LIBRARY_PATH=./lib ./bin/flutter ./build/flutter_assets/
icu_data_path: /home/pi/bin/icudtl.dat
assets_path: ./build/flutter_assets/
GLFW error 65544: X11: RandR gamma ramp support seems broken
flutter: Observatory listening on http://127.0.0.1:45527/OMa0lhVaBWo=/
../../third_party/dart/runtime/vm/object.h: 3038: error: Handle check failed: saw 0 expected Function
Aborted
```

DisDis, 2019-06-07:
@drewcovi what for do you use sudo?

drewcovi, 2019-06-07:
im running on raspbian lite, so i fired up `sudo startx` but perhaps i didnt need to

drewcovi, 2019-06-07:
Based on the symlinks you have above, it seems we need a few other dependencies before building sysroot `libx11-dev libgtk-3-dev libglfw3`

DisDis, 2019-06-07:
@drewcovi pls install 'Raspbian Stretch with desktop', and use run.sh(flutter_armv7.tar.xz)

drewcovi, 2019-06-07:
for sure! thanks @DisDis. I'll take a pass later on. the instructions on this repo talk about checking out a lite version of raspbian for sysroot which is why i used lite.

DisDis, 2019-06-07:
@drewcovi everything has changed:). no performance anyway. Flutter is very slow on RasPi3.

drewcovi, 2019-06-07:
Sigh... then perhaps I'm back to WPE and web application frameworks.... I can close this for now, as it seems there's a path forward building in Ubuntu. Have a good weekend!

closed 2019-06-07 by drewcovi

pauldemarco, 2019-06-18:
I am also seeing this:

> /home/paul/flutter_embedded/sdk/toolchain/bin/arm-linux-gnueabihf-ld: unrecognized option '--color-diagnostics'
/home/paul/flutter_embedded/sdk/toolchain/bin/arm-linux-gnueabihf-ld: use the --help option for usage information
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)

Should this issue be re-opened, what is the solution?