meta-flutter/flutter_embedded #16: Error building due to --color-diagnostics

pauldemarco, 2019-06-18:
> /home/paul/flutter_embedded/sdk/toolchain/bin/arm-linux-gnueabihf-ld: unrecognized option '--color-diagnostics'
/home/paul/flutter_embedded/sdk/toolchain/bin/arm-linux-gnueabihf-ld: use the --help option for usage information
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)

Possibly related to #13

DisDis, 2019-06-18:
https://github.com/jwinarske/flutter_embedded/issues/13#issuecomment-499687915

pauldemarco, 2019-06-18:
Thanks, whats the best way to get this baked in?

DisDis, 2019-06-18:
@pauldemarco 
1) Change it.
2) Remove build/toolchain-prefix/src/toolchain-stamp/toolchain-build and toolchain-configure
3) autoninja

DisDis, 2019-06-18:
or change files manually.
use: grep -iRl "color-diagnostics" ./build/

pauldemarco, 2019-06-19:
@DisDis 

Sorry, let me clarify. By "baked in" I mean so we don't have to do any additional steps outside of the build commands.

DisDis, 2019-06-19:
@pauldemarco 
i wrote all steps.
1. please change 'LINKER_SUPPORTS_COLOR_DIAGNOSTICS:INTERNAL=0'
2. delete two files.
3. run autoninja

pauldemarco, 2019-06-19:
Great, I'll go ahead and see if I can bake this in with a PR. 

Thanks!

closed 2019-06-19 by pauldemarco

DisDis, 2019-06-19:
@pauldemarco for PR you only need to change 'LINKER_SUPPORTS_COLOR_DIAGNOSTICS:INTERNAL=0'