meta-flutter/flutter_embedded #18: lubflutter_linux.so not created

drewcovi, 2019-06-26:
Following the build process without mods (except for the --color-diagnostics flag removal).

No issues besides those two hiccups. Any reason libflutter_linux.so wouldn't exist?

```
Error copying file "/path/to/flutter_embedded/build/engine-prefix/src/engine/src/out/linux_dynamic_release_arm/libflutter_linux.so" to "/path/to/flutter_embedded/build/target/lib".
```

```
-rw-rw-r-- 2 drew drew  36K Jun 25 12:54 flutter_embedder.h
-rw-rw-r-- 3 drew drew 9.8M Jun 25 12:55 icudtl.dat
-rw-rw-r-- 1 drew drew 1.1K Jun 25 12:59 args.gn
-rw-rw-r-- 1 drew drew 208K Jun 25 12:59 toolchain.ninja
-rw-rw-r-- 1 drew drew 8.2K Jun 25 12:59 build.ninja.d
-rw-rw-r-- 1 drew drew  71K Jun 25 12:59 build.ninja
drwxrwxr-x 3 drew drew 4.0K Jun 25 12:59 ..
drwx------ 5 drew drew 4.0K Jun 25 12:59 gen
-rw-rw-r-- 1 drew drew 795K Jun 25 13:00 vm_outline_strong.dill
-rw-rw-r-- 1 drew drew 5.4M Jun 25 13:00 vm_platform_strong.dill
-rw-rw-r-- 1 drew drew  68K Jun 25 13:00 vm_platform_strong.dill.d
drwxrwxr-x 2 drew drew 4.0K Jun 25 13:03 flutter_patched_sdk
drwx------ 5 drew drew 4.0K Jun 25 13:10 clang_x86
-rw-rw-r-- 1 drew drew 1.6M Jun 25 13:11 .ninja_deps
drwxrwxr-x 2 drew drew 4.0K Jun 25 13:11 so.unstripped
-rw-rw-r-- 1 drew drew  30K Jun 25 13:11 libflutter_engine.so.TOC
-rwxrwxr-x 1 drew drew  21M Jun 25 13:11 libflutter_engine.so
drwx------ 7 drew drew 4.0K Jun 25 13:11 .
drwx------ 5 drew drew 4.0K Jun 25 13:11 obj
-rw-rw-r-- 1 drew drew 391K Jun 25 13:11 .ninja_log
```

jwinarske, 2019-07-02:
I spun up a build before leaving work today.  The only parameter change was for Clang 8.0.0.  Aside from that the only build break I hit was related to parallel building.  What's your OS + build environment?

Aside from not seeing libflutter_linux.so, have you checked the so dependencies for your RPI executable?  Use `readelf -d "so/executable name"`

drewcovi, 2019-07-02:
Ubuntu Server 18.04.2 LTS, otherwise simply following the guide.... i can try again though...

drewcovi, 2019-07-02:
Any recommendations on ENV variables to avoid the `color-diagnostics` bug?

jwinarske, 2019-07-03:
I haven't hit that.  I open a terminal, no unique exports, run the CMake sequence for Unix makefile (default), then make.  Did you run the shell script from the repo to install dependencies?  I modified it for Bionic.

You might also consider trying to build in a Docker container.  I'll take a look at making a Docker image script in a week or so.

drewcovi, 2019-07-09:
I did run the shell script, but I'm also running headless ubuntu server... so its possible my system simply doesnt support color-diagnostics

jwinarske, 2020-04-27:
hi @drewcovi since this is still open you might trying building with master.  I pulled out the external clang dependencies, and use the Fuchsia Clang 11.0 (CIPD).  Once I have tested on ubuntu server, I'll close this.

jwinarske, 2020-05-03:
Tested.  Closing.

closed 2020-05-03 by jwinarske