meta-flutter/flutter_embedded #19: Touchscreen not working

poky, 2019-07-07:
Hi, I have made it to run in CLI mode with the following outputs:
It's on Raspberry Pi 3B with Official 7" Touchscreen
I can confirm that the touchscreen works under startx, and if I run flutter under startx, the touch can work, but not under pure CLI mode, can someone advise, thanks!

`LOG: ../flutter/main.cc:66: Display Size: 800 x 480
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(95)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter: Observatory listening on http://127.0.0.1:41495/ugnN4EAk8o4=/
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(95)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
[ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: Exception: operation failed
#0      ShaderWarmUp.execute (package:flutter/src/painting/shader_warm_up.dart:90:5)
<asynchronous suspension>
#1      _WidgetsFlutterBinding&BindingBase&GestureBinding&ServicesBinding&SchedulerBinding&PaintingBinding.initInstances (package:flutter/src/painting/binding.dart:25:20)
#2      _WidgetsFlutterBinding&BindingBase&GestureBinding&ServicesBinding&SchedulerBinding&PaintingBinding&SemanticsBinding.initInstances (package:flutter/src/semantics/binding.dart:22:11)
#3      _WidgetsFlutterBinding&BindingBase&GestureBinding&ServicesBinding&SchedulerBinding&PaintingBinding&SemanticsBinding&RendererBinding.initInstances (package:flutter/src/rendering/binding.dart:29:11)
#4      _WidgetsFlutterBinding&BindingBase&GestureBinding&ServicesBinding&SchedulerBinding&PaintingBinding&SemanticsBinding&RendererBinding&WidgetsBinding.initInstances (package:flutter/src/widgets/binding.dart:252:11)
#5      new BindingBase (package:flutter/src/foundation/binding.dart:56:5)
#6      new _WidgetsFlutterBinding&BindingBase&GestureBinding (package:flutter/src/widgets/binding.dart)
#7      new _WidgetsFlutterBinding&BindingBase&GestureBinding&ServicesBinding (package:flutter/src/widgets/binding.dart)
#8      new _WidgetsFlutterBinding&BindingBase&GestureBinding&ServicesBinding&SchedulerBinding (package:flutter/src/widgets/binding.dart)
#9      new _WidgetsFlutterBinding&BindingBase&GestureBinding&ServicesBinding&SchedulerBinding&PaintingBinding (package:flutter/src/widgets/binding.dart)
#10     new _WidgetsFlutterBinding&BindingBase&GestureBinding&ServicesBinding&SchedulerBinding&PaintingBinding&SemanticsBinding (package:flutter/src/widgets/binding.dart)
#11     new _WidgetsFlutterBinding&BindingBase&GestureBinding&ServicesBinding&SchedulerBinding&PaintingBinding&SemanticsBinding&RendererBinding (package:flutter/src/widgets/binding.dart)
#12     new _WidgetsFlutterBinding&BindingBase&GestureBinding&ServicesBinding&SchedulerBinding&PaintingBinding&SemanticsBinding&RendererBinding&WidgetsBinding (package:flutter/src/widgets/binding.dart)
#13     new WidgetsFlutterBinding (package:flutter/src/widgets/binding.dart)
#14     WidgetsFlutterBinding.ensureInitialized (package:flutter/src/widgets/binding.dart:994:7)
#15     runApp (package:flutter/src/widgets/binding.dart:785:25)
#16     main (package:example_flutter/main.dart:23:3)
#17     _runMainZoned.<anonymous closure>.<anonymous closure> (dart:ui/hooks.dart:216:25)
#18     _rootRun (dart:async/zone.dart:1124:13)
#19     _CustomZone.run (dart:async/zone.dart:1021:19)
#20     _runZoned (dart:async/zone.dart:1516:10)
#21     runZoned (dart:async/zone.dart:1500:12)
#22     _runMainZoned.<anonymous closure> (dart:ui/hooks.dart:208:5)
#23     _startIsolate.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:305:19)
#24     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:172:12)
`

jwinarske, 2019-07-08:
Everything works short of touch?

poky, 2019-07-09:
> Everything works short of touch?

Here's what I have so far, I can get the UI to display without the desktop, but no touch screen so I can't test beyond this point.

[Youtube link](https://www.youtube.com/watch?v=3hmxaH_z7mY)

jwinarske, 2019-07-10:
I suspect the touch became broken after changes to support Linux desktop.  @disdis can you take a look at this?

DisDis, 2019-07-11:
Yes, i broken touch. I can fix it. No reason to fix it. Flutter works very slowly on RaPi3(1-3fps). Touching is  not solve speed problem. 
I tried to rebuild Flutter with fix https://github.com/flutter/flutter/issues/30730 . The speed is low, the processor is 100% loaded

ardera, 2019-07-19:
@DisDis The slow speed definitely isn't RaPi3's fault. Got steady 60fps on my own embedder. Maybe you specified some "wrong" build flags?

poky, 2019-07-20:
@ardera Can you advise how to fix it? All I want is to see how Flutter runs under CLI with touchscreen support, thanks!

jwinarske, 2020-04-27:
@poky I started up on building a Wayland shell.  You might try building the Flutter Engine with desktop enabled.  This has the basics working and is x11 based.  Just a matter of getting the build flags right.

closed 2020-05-03 by jwinarske