meta-flutter/flutter_embedded #20: Build failed in rpi_flutter step

poky, 2019-07-09:
Hi, I'm using the latest Raspbian Buster Lite (2019-06-20) for the build, however it stopped with the following messages:

```
[1/3] Performing build step for 'rpi_flutter'
FAILED: rpi_flutter-prefix/src/rpi_flutter-stamp/rpi_flutter-build 
cd /home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter-build && /usr/bin/cmake --build . && /usr/bin/cmake -E touch /home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter-stamp/rpi_flutter-build
[1/1] Linking CXX executable flutter
FAILED: flutter 
: && /home/lee/flutter_embedded/sdk/toolchain/bin/clang++ --target=arm-linux-gnueabihf --sysroot=/home/lee/flutter_embedded/sdk/sysroot  -stdlib=libc++ -I/home/lee/flutter_embedded/third_party/libcxx/include -U_FORTIFY_SOURCE -Wall -g -ftree-vectorize -pipe -O3 -DNDEBUG --rtlib=compiler-rt -fuse-ld=gold -lc++abi -lunwind  CMakeFiles/flutter.dir/flutter/main.cc.o CMakeFiles/flutter.dir/flutter/utils.cc.o CMakeFiles/flutter.dir/flutter/cpp_client_wrapper/engine_method_result.cc.o CMakeFiles/flutter.dir/flutter/cpp_client_wrapper/flutter_window_controller.cc.o CMakeFiles/flutter.dir/flutter/cpp_client_wrapper/json_message_codec.cc.o CMakeFiles/flutter.dir/flutter/cpp_client_wrapper/plugin_registrar.cc.o CMakeFiles/flutter.dir/flutter/cpp_client_wrapper/standard_codec.cc.o  -o flutter -L/home/lee/flutter_embedded/sdk/sysroot/opt/vc/lib  -L/home/lee/flutter_embedded/build/engine-prefix/src/engine/src/out/linux_dynamic_release_arm -Wl,-rpath,/opt/vc/lib:/home/lee/flutter_embedded/build/engine-prefix/src/engine/src/out/linux_dynamic_release_arm: -lts -lbrcmGLESv2 -lbrcmEGL -lbcm_host -lflutter_engine -lpthread -ldl -lflutter_linux && :
/home/lee/flutter_embedded/sdk/toolchain/bin/arm-linux-gnueabihf-ld.gold: error: cannot find -lflutter_linux
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/flutter_window_controller.cc:15: error: undefined reference to 'FlutterDesktopInit'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/flutter_window_controller.cc:20: error: undefined reference to 'FlutterDesktopDestroyWindow'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/flutter_window_controller.cc:23: error: undefined reference to 'FlutterDesktopTerminate'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/flutter_window_controller.cc:50: error: undefined reference to 'FlutterDesktopCreateWindow'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/flutter_window_controller.cc:58: error: undefined reference to 'FlutterDesktopGetWindow'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/flutter_window_controller.cc:70: error: undefined reference to 'FlutterDesktopGetPluginRegistrar'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/flutter_window_controller.cc:75: error: undefined reference to 'FlutterDesktopRunWindowLoop'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/plugin_registrar.cc:86: error: undefined reference to 'FlutterDesktopMessengerSend'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/plugin_registrar.cc:94: error: undefined reference to 'FlutterDesktopMessengerSetCallback'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/plugin_registrar.cc:102: error: undefined reference to 'FlutterDesktopMessengerSetCallback'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/plugin_registrar.cc:110: error: undefined reference to 'FlutterDesktopRegistrarGetMessenger'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/plugin_registrar.cc:122: error: undefined reference to 'FlutterDesktopRegistrarEnableInputBlocking'
/home/lee/flutter_embedded/build/rpi_flutter-prefix/src/rpi_flutter/flutter/cpp_client_wrapper/plugin_registrar.cc:36: error: undefined reference to 'FlutterDesktopMessengerSendResponse'
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.

```

DisDis, 2019-07-11:
Last flutter is not build.
I have fix, i will try to PR

DisDis, 2019-07-11:
https://github.com/jwinarske/flutter_embedded/pull/21

poky, 2019-07-12:
Hi, just tried the new updated one, came with this error, could you tell me how to fix this, thanks!

```
[24/36] Performing patch step for 'engine'
FAILED: engine-prefix/src/engine-stamp/engine-patch 
cd /home/lee/flutter_embedded/build/engine-prefix/src/engine && cd /home/lee/flutter_embedded/build/engine-prefix/src/engine && cd src && git checkout build/config/compiler/BUILD.gn && git apply /home/lee/flutter_embedded/build/engine_compiler_build.patch && cd flutter && git apply /home/lee/flutter_embedded/cmake/patches/engine.patch && cd .. && cd third_party/dart && git checkout runtime/BUILD.gn && git apply /home/lee/flutter_embedded/cmake/patches/dart.patch && cd ../../.. && /usr/bin/cmake -E touch /home/lee/flutter_embedded/build/engine-prefix/src/engine-stamp/engine-patch
error: unrecognized input
ninja: build stopped: subcommand failed.
```

DisDis, 2019-07-12:
@poky i will look at night. Currently you can try clean install.(remove all, git clone, rebuild all)

poky, 2019-07-12:
@DisDis Yes, I did, complete new git clone, cmake and autoninja
Maybe have something to do with the empty engine.patch file?

poky, 2019-07-13:
@DisDis I manually removed the engine-patch in the build.ninja to bypass the above errors, however it continued with these int errors when building libcxx

```
FAILED: obj/third_party/libcxx/src/libcxx.chrono.o 
/home/lee/flutter_embedded/sdk/toolchain/bin/clang++ -MMD -MF obj/third_party/libcxx/src/libcxx.chrono.o.d --target=arm-linux-gnueabihf --sysroot /home/lee/flutter_embedded/sdk/sysroot -D_LIBCPP_NO_EXCEPTIONS -D_LIBCPP_NO_RTTI -D_LIBCPP_BUILDING_LIBRARY -DLIBCXX_BUILDING_LIBCXXABI -DUSE_OPENSSL=1 -DTOOLCHAIN_VERSION=bac220c15490dcf7b7d8136f75100bbc77e8d217 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -I../../third_party/libcxx/include -I../../third_party/libcxxabi/include -fno-strict-aliasing -fstack-protector --param=ssp-buffer-size=4 -march=armv7-a -mfloat-abi=hard -mtune=generic-armv7-a -mthumb -fPIC -pipe -pthread -fcolor-diagnostics -mfpu=neon -Wall -Wextra -Wendif-labels -Werror -Wno-missing-field-initializers -Wno-unused-parameter -fvisibility=hidden --sysroot=/home/lee/flutter_embedded/sdk/sysroot -Wheader-hygiene -Wstring-conversion -Wthread-safety -O2 -fno-ident -fdata-sections -ffunction-sections -g0 -Wno-thread-safety-analysis -fvisibility-inlines-hidden -std=c++17 -fno-rtti -stdlib=libc++ -I/home/lee/flutter_embedded/third_party/libcxx/include -I/home/lee/flutter_embedded/third_party/libcxxabi/include -fno-exceptions -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -fno-exceptions -c ../../third_party/libcxx/src/chrono.cpp -o obj/third_party/libcxx/src/libcxx.chrono.o
In file included from ../../third_party/libcxx/src/chrono.cpp:10:
In file included from ../../third_party/libcxx/include/chrono:800:
In file included from ../../third_party/libcxx/include/ratio:82:
../../third_party/libcxx/include/cstdint:153:8: error: no member named 'int8_t' in the global namespace; did you mean '__int8_t'?
using::int8_t;
     ~~^
/home/lee/flutter_embedded/sdk/sysroot/usr/include/arm-linux-gnueabihf/bits/types.h:36:21: note: '__int8_t' declared here
typedef signed char __int8_t;
                    ^
In file included from ../../third_party/libcxx/src/chrono.cpp:10:
In file included from ../../third_party/libcxx/include/chrono:800:
In file included from ../../third_party/libcxx/include/ratio:82:
../../third_party/libcxx/include/cstdint:154:8: error: no member named 'int16_t' in the global namespace; did you mean '__int16_t'?
using::int16_t;
     ~~^
/home/lee/flutter_embedded/sdk/sysroot/usr/include/arm-linux-gnueabihf/bits/types.h:38:26: note: '__int16_t' declared here
typedef signed short int __int16_t;
                         ^
In file included from ../../third_party/libcxx/src/chrono.cpp:10:
In file included from ../../third_party/libcxx/include/chrono:800:
In file included from ../../third_party/libcxx/include/ratio:82:
../../third_party/libcxx/include/cstdint:155:8: error: no member named 'int32_t' in the global namespace; did you mean '__int32_t'?
using::int32_t;
     ~~^
/home/lee/flutter_embedded/sdk/sysroot/usr/include/arm-linux-gnueabihf/bits/types.h:40:20: note: '__int32_t' declared here
typedef signed int __int32_t;
                   ^
In file included from ../../third_party/libcxx/src/chrono.cpp:10:
In file included from ../../third_party/libcxx/include/chrono:800:
In file included from ../../third_party/libcxx/include/ratio:82:
../../third_party/libcxx/include/cstdint:156:8: error: no member named 'int64_t' in the global namespace; did you mean '__int64_t'?
using::int64_t;
     ~~^
/home/lee/flutter_embedded/sdk/sysroot/usr/include/arm-linux-gnueabihf/bits/types.h:46:44: note: '__int64_t' declared here
__extension__ typedef signed long long int __int64_t;
                                           ^
In file included from ../../third_party/libcxx/src/chrono.cpp:10:
In file included from ../../third_party/libcxx/include/chrono:800:
In file included from ../../third_party/libcxx/include/ratio:82:
../../third_party/libcxx/include/cstdint:158:8: error: no member named 'uint8_t' in the global namespace
using::uint8_t;
     ~~^
../../third_party/libcxx/include/cstdint:159:8: error: no member named 'uint16_t' in the global namespace
using::uint16_t;
     ~~^
../../third_party/libcxx/include/cstdint:160:8: error: no member named 'uint32_t' in the global namespace
using::uint32_t;
     ~~^
../../third_party/libcxx/include/cstdint:161:8: error: no member named 'uint64_t' in the global namespace
using::uint64_t;
     ~~^
../../third_party/libcxx/include/cstdint:163:8: error: no member named 'int_least8_t' in the global namespace
using::int_least8_t;
     ~~^
../../third_party/libcxx/include/cstdint:164:8: error: no member named 'int_least16_t' in the global namespace
using::int_least16_t;
     ~~^
../../third_party/libcxx/include/cstdint:165:8: error: no member named 'int_least32_t' in the global namespace
using::int_least32_t;
     ~~^
../../third_party/libcxx/include/cstdint:166:8: error: no member named 'int_least64_t' in the global namespace
using::int_least64_t;
     ~~^
../../third_party/libcxx/include/cstdint:168:8: error: no member named 'uint_least8_t' in the global namespace
using::uint_least8_t;
[... 105 lines cut ...]
     ~~^
[... cut ...]
[484/3386] CXX obj/third_party/dart/runtime/platform/libdart_platform_jit.floating_point_win.o
[485/3386] CXX obj/third_party/dart/runtime/platform/libdart_platform_jit.assert.o
[486/3386] CXX obj/third_party/dart/runtime/platform/libdart_platform_jit.text_buffer.o
[487/3386] CXX obj/third_party/dart/runtime/bin/libdart_builtin.file.o
[488/3386] CC obj/third_party/boringssl/src/crypto/fipsmodule/crypto.bcm.o
[489/3386] ACTION //third_party/dart/runtime/observatory:build_observatory(//build/toolchain/custom:custom)
ninja: build stopped: subcommand failed.
```

jwinarske, 2020-04-27:
Hi @poky.  If you're interested in building Tip Of Tree Flutter Engine, what I have in master currently does this.  I eliminated the external Clang dependency and use the Fuchsia Clang CIPD package.

closed 2020-05-03 by jwinarske