meta-flutter/flutter_embedded #26: gclient.py: error: Inconsistent arguments. Use either --spec or one or 2 args

skinkie, 2020-05-02:
I have checked out the latest depot_tools.
```
export PATH=$PATH:/home/skinkie/Sources/depot_tools
```
In order to prevent the Python3 warning:
```
export GCLIENT_PY3=0
```
```
make
[  4%] Performing download step for 'engine'
Usage: gclient.py config [options] [url]

gclient.py: error: Inconsistent arguments. Use either --spec or one or 2 args
make[2]: *** [CMakeFiles/engine.dir/build.make:110: engine-prefix/src/engine-stamp/engine-download] Error 2
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/engine.dir/all] Error 2
make: *** [Makefile:104: all] Error 2
```

skinkie, 2020-05-02:
When I perform the following step manually, and by adding quotes, it does work.

gclient config --spec **"**solutions=[{\"managed\":False,\"name\":\"src/flutter\",\"url\":\"git@github.com:flutter/engine.git\",\"custom_vars\":{\"download_android_deps\":False,\"download_windows_deps\":False,}}]**"**

jwinarske, 2020-05-02:
Hi @skinkie.  What host are you building with?

The pattern I used with Ubuntu 18.0.4 LTS: 
1. update path to include depot_tools
2. `mkdir build && cd build`
3. `cmake ..`
4. `make -j VERBOSE=1`

skinkie, 2020-05-02:
I am trynig to build on gentoo, as you see in the rest of the issues that fails big time :) But I obviously can test it on a ubuntu system. Specifically regarding the fontconfig part: is it expected to install the ARM package?

Is building the flutter-engine directly on a pi "easier"?

jwinarske, 2020-05-02:
The stuff you're hitting look straight forward.  What version of Python?  If three, you might want to add 2.7 so it shows up earlier in your path.  Otherwise the other way around.

I'm not sure the Clang package is setup for building Native = ARM, so it may be a host-only exercise.

Be sure your PKG_CONFIG_PATH enviromental variable is pointing to the sysroot pkgconfig folder which has the fontconfig pc file.  The other option is to remove the enable fontconfig flag to make forward progress. It is needed to render text though.

skinkie, 2020-05-02:
python2 and python3 are installed, using eselect I have selected 2.7. This still didn't work for me.

jwinarske, 2020-05-02:
I would try to a host build of the Flutter engine without CMake to sort out your environment.  Once you have that working, then try with CMake.

closed 2020-05-03 by jwinarske