meta-flutter/flutter_embedded #27: SyntaxError: Missing parentheses in call to 'print'.

skinkie, 2020-05-02:
This feels like a python2 vs python3 problem. 
```
[ 16%] Performing configure step for 'engine'
  File "./flutter/tools/gn", line 395
    print "Generating GN files in: %s" % out_dir
                                     ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("Generating GN files in: %s" % out_dir)?
make[2]: *** [CMakeFiles/engine.dir/build.make:127: engine-prefix/src/engine-stamp/engine-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/engine.dir/all] Error 2
make: *** [Makefile:104: all] Error 2
```
While python2 is aliased and the variable is set.

skinkie, 2020-05-02:
Exactly the same issue on Ubuntu. Changing /usr/bin/python exclitly to /usr/bin/python2 brings me back to the pkg-config error...

jwinarske, 2020-05-02:
pkg config is related to not having a valid PKG_CONFIG_PATH, mostly null.  Build will fail as it cannot find the dependency.  If you disable the enable fontconfig flag, it will ignore the package and carry on.

closed 2020-05-03 by jwinarske