meta-flutter/flutter_embedded #28: subprocess.CalledProcessError: Command '['pkg-config', '--variable=prefix', 'freetype2']' returned non-zero exit status 1

skinkie, 2020-05-02:
```
python /mnt/storage/home/skinkie/Sources/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py -s /home/skinkie/Sources/flutter_embedded/sysroot -a arm --system_libdir lib freetype2
Traceback (most recent call last):
  File "/mnt/storage/home/skinkie/Sources/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py", line 249, in <module>
    sys.exit(main())
  File "/mnt/storage/home/skinkie/Sources/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py", line 144, in main
    prefix = GetPkgConfigPrefixToStrip(options, args)
  File "/mnt/storage/home/skinkie/Sources/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py", line 82, in GetPkgConfigPrefixToStrip
    "--variable=prefix"] + args, env=os.environ).decode('utf-8')
  File "/usr/lib64/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['pkg-config', '--variable=prefix', 'freetype2']' returned non-zero exit status 1
```

jwinarske, 2020-05-02:
`sudo apt-get install libfreetype6-dev`

skinkie, 2020-05-02:
I am not using ubuntu. But the pkgconfig file exists and can be found using the command;
```
skinkie@thinkpad ~ $ ls -l /usr/lib64/pkgconfig/freetype2.pc 
-rw-r--r-- 1 root root 321 Feb  8 14:49 /usr/lib64/pkgconfig/freetype2.pc
skinkie@thinkpad ~ $ pkg-config --variable=prefix freetype2
/usr
```
So I do not understand why the error states that the returncode is non-zero.

skinkie, 2020-05-02:
This is the ubuntu output:

```
sudo apt-get install libfreetype6-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libfreetype6-dev is already the newest version (2.9.1-4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
mio@mio-Aspire-5750G:~/Sources/flutter_embedded/build$ make
[  4%] Performing configure step for 'engine'
Generating GN files in: out/linux_debug_arm
ERROR at //build/config/linux/pkg_config.gni:103:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /home/mio/Sources/flutter_embedded/third_party/engine/src/out/linux_debug_arm/
Command: python /home/mio/Sources/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py -s /home/mio/Sources/flutter_embedded/sysroot -a arm --system_libdir lib freetype2
Returned 1.
stderr:

Package freetype2 was not found in the pkg-config search path.
Perhaps you should add the directory containing `freetype2.pc'
to the PKG_CONFIG_PATH environment variable
No package 'freetype2' found
Traceback (most recent call last):
  File "/home/mio/Sources/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py", line 248, in <module>
    sys.exit(main())
  File "/home/mio/Sources/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py", line 143, in main
    prefix = GetPkgConfigPrefixToStrip(options, args)
  File "/home/mio/Sources/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py", line 82, in GetPkgConfigPrefixToStrip
    "--variable=prefix"] + args, env=os.environ).decode('utf-8')
  File "/usr/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['pkg-config', '--variable=prefix', 'freetype2']' returned non-zero exit status 1

See //build/config/linux/BUILD.gn:30:1: whence it was called.
pkg_config("freetype2") {
^------------------------
See //build/config/BUILDCONFIG.gn:352:33: which caused the file to be included.
  _native_compiler_configs += [ "//build/config/linux:sdk" ]
                                ^-------------------------
make[2]: *** [CMakeFiles/engine.dir/build.make:106: engine-prefix/src/engine-stamp/engine-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/engine.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

jwinarske, 2020-05-02:
Your setup is different than the default.  You need this:
`export PKG_CONFIG_PATH=/usr/lib64/pkgconfig`

skinkie, 2020-05-02:
@jwinarske the output above is a regular xubuntu image. But happy to try it with that variable.

skinkie, 2020-05-02:
With the export I still see:
```
subprocess.CalledProcessError: Command '['pkg-config', '--variable=prefix', 'freetype2']' returned non-zero exit status 1

See //build/config/linux/BUILD.gn:30:1: whence it was called.
pkg_config("freetype2") {
^------------------------
See //build/config/BUILDCONFIG.gn:352:33: which caused the file to be included.
  _native_compiler_configs += [ "//build/config/linux:sdk" ]
                                ^-------------------------
make[2]: *** [CMakeFiles/engine.dir/build.make:106: engine-prefix/src/engine-stamp/engine-configure] Fout 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/engine.dir/all] Fout 2
make: *** [Makefile:84: all] Fout 2
mio@mio-Aspire-5750G:~/Sources/flutter_embedded/build$ pkg-config --variable=prefix freetype2
/usr
```

jwinarske, 2020-05-02:
I say step one is build an Engine for host, without CMake.  This will be the minimum problem set best suited to flush out the environment.

skinkie, 2020-05-02:
@jwinarske and here you mean just a step by step https://github.com/flutter/flutter/wiki/Compiling-the-engine#compiling-for-macos-or-linux ?

skinkie, 2020-05-02:
I can confirm (on gentoo) that building with python2 ./flutter/tools/gn --runtime-mode  release  --target-os linux   --linux-cpu x64 --lto --clang    --enable-fontconfig  --enable-skshaper   --embedder-for-target --stripped gave me output, now going to see if it gives me something on screen ;)

jwinarske, 2020-05-17:
Is this still outstanding?

jwinarske, 2020-05-19:
Closing since this has been resolved

closed 2020-05-19 by jwinarske