meta-flutter/flutter_embedded #29: Linker error

ranciere, 2020-05-04:
I've just checked out the repo and build with defaults (cmake.. & make). The linker error is:
`SOLINK ./libflutter_engine.so
FAILED: libflutter_engine.so so.unstripped/libflutter_engine.so libflutter_engine.so.TOC 
/home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/clang++ --target=armv7-unknown-linux-gnueabihf --sysroot /home/ranciere/work/flutter_embedded/sysroot -shared -Wl,--fatal-warnings -fPIC -Wl,-z,noexecstack -Wl,-z,now -Wl,-z,relro -Wl,-z,defs -pthread --sysroot=/home/ranciere/work/flutter_embedded/sysroot  -Wl,-O2 -Wl,--gc-sections -Wl,--as-needed -o ./so.unstripped/libflutter_engine.so  -Wl,--build-id=sha1 -Wl,-soname=libflutter_engine.so @./libflutter_engine.so.rsp && { /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/llvm-readelf -d ./so.unstripped/libflutter_engine.so | grep SONAME ; /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/llvm-nm -gD -f posix ./so.unstripped/libflutter_engine.so | cut -f1-2 -d' '; } > ./libflutter_engine.so.tmp && if ! cmp -s ./libflutter_engine.so.tmp ./libflutter_engine.so.TOC; then mv ./libflutter_engine.so.tmp ./libflutter_engine.so.TOC; fi && /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/llvm-strip -o ./libflutter_engine.so ./so.unstripped/libflutter_engine.so
ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by allocatestack.c:145
>>>               pthread_create.o:(stack_list_del) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by allocatestack.c:282
>>>               pthread_create.o:(free_stacks) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by allocatestack.c:282
>>>               pthread_create.o:(free_stacks) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by allocatestack.c:160
>>>               pthread_create.o:(stack_list_add) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against symbol: start_thread in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by createthread.c:142 (../sysdeps/unix/sysv/linux/createthread.c:142)
>>>               pthread_create.o:(create_thread) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by createthread.c:142 (../sysdeps/unix/sysv/linux/createthread.c:142)
>>>               pthread_create.o:(create_thread) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by createthread.c:142 (../sysdeps/unix/sysv/linux/createthread.c:142)
>>>               pthread_create.o:(create_thread) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by createthread.c:142 (../sysdeps/unix/sysv/linux/createthread.c:142)
>>>               pthread_create.o:(create_thread) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by futex-internal.h:200 (../sysdeps/nptl/futex-internal.h:200)
>>>               pthread_create.o:(setxid_mark_thread.isra.1) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by allocatestack.c:1059
>>>               pthread_create.o:(setxid_unmark_thread.isra.3.part.4) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against symbol: __stack_chk_guard in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/lib/arm-linux-gnueabihf/ld-linux-armhf.so.3
>>> referenced by allocatestack.c:1090
>>>               pthread_create.o:(setxid_signal_thread) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against symbol: __stack_chk_guard in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/lib/arm-linux-gnueabihf/ld-linux-armhf.so.3
>>> referenced by allocatestack.c:1111
[... cut ...]
>>> defined in /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by allocatestack.c:888
>>>               pthread_create.o:(__reclaim_stacks) in archive /home/ranciere/work/flutter_embedded/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
clang-11: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
`

jwinarske, 2020-05-05:
Interesting.  Did the patches take?  

Maybe a log of the configure and complete build?

ranciere, 2020-05-05:
It seems it is an user error: I've rerun it from scratch, and now there is no error. I close the ticket!

closed 2020-05-05 by ranciere