meta-flutter/flutter_embedded #30: Missing flutter_linux library

ranciere, 2020-05-11:
I've compiled it via the _cmake .. -DBUILD_FLUTTER_RPI=On_ then I've got the following error:
`ranciere@ranciere:~/work/flutter_embedded/build$ make
[ 16%] Built target engine
[ 18%] Performing build step for 'rpi_flutter'
[ 12%] Linking CXX executable flutter
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lflutter_linux
collect2: error: ld returned 1 exit status
make[5]: *** [CMakeFiles/flutter.dir/build.make:174: flutter] Error 1
make[4]: *** [CMakeFiles/Makefile2:76: CMakeFiles/flutter.dir/all] Error 2
make[3]: *** [Makefile:130: all] Error 2
make[2]: *** [CMakeFiles/rpi_flutter.dir/build.make:113: rpi_flutter-prefix/src/rpi_flutter-stamp/rpi_flutter-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:86: CMakeFiles/rpi_flutter.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
`
How can I build this lib?

jwinarske, 2020-05-11:
@ranciere I'll take a look.  What type of shell are you looking for?  Just anything that runs on RPI3?

ranciere, 2020-05-11:
Just anything that runs on _RPI4_ :) First I would like to run something on my PI in interpreter mode, but later it would bo be good to run an AOT application, debug, hot reload etc.

jwinarske, 2020-05-11:
Okay.  Give me a few days too take a look.  Are you looking for full screen custom, or just standard flutter desktop support?

jwinarske, 2020-05-12:
So if you're looking for Flutter Engine Linux desktop support, you should have most of what you need.  You should build with my changes in master.  Something like this:
`cmake .. -DBUILD_FLUTTER_RPI=ON -UENGINE_EMBEDDER_FOR_TARGET -UENGINE_DISABLE_DESKTOP -DTARGET_HOSTNAME=mydevice`
It needs to sync the sysroot from the target, and the target needs all the dependencies installed prior.  gtk3+, etc.  I haven't worked through what they are yet.  Let me know what you find.

ranciere, 2020-05-12:
I've tried with the newest version. Running just 'cmake ..' results:
```
[ 60%] Performing configure step for 'engine'
Generating GN files in: out/linux_debug_arm
ERROR at //build/config/linux/pkg_config.gni:103:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /home/ranciere/work/flutter_embedded/third_party/engine/src/out/linux_debug_arm/
Command: python /home/ranciere/work/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py -s /home/ranciere/work/flutter_embedded/sysroot -a arm --system_libdir lib gtk+-3.0
Returned 1.
stderr:

Package gtk+-3.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gtk+-3.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gtk+-3.0' found

```

ranciere, 2020-05-12:
Interesting: the compiling of flutter engine runs without compile error, but the copying is unsuccesfull:
```
[ 56%] Performing install step for 'engine'
Error copying file "out/linux_debug_arm/libflutter_engine.so" to "/home/ranciere/work/flutter_embedded/build".
make[2]: *** [CMakeFiles/engine.dir/build.make:74: engine-prefix/src/engine-stamp/engine-install] Error 1
make[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/engine.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```
It's a naming error, the name of the lib is libflutter_engine_**gtk**.so, not libflutter_engine.so

ranciere, 2020-05-12:
With ```cmake .. -DBUILD_FLUTTER_RPI=ON -DTARGET_SYSROOT=/home/ranciere/work/sysroot```, I will get: 
```
ranciere@ranciere:~/work/flutter_embedded/buil$ make -j8
[ 16%] Built target depot_tools
[ 33%] Built target sysroot
[ 50%] Built target symlink_fixups
[ 52%] Performing build step for 'engine'
ninja: Entering directory `out/linux_debug_arm'
[1/1] Regenerating ninja files
FAILED: build.ninja 
../../flutter/third_party/gn/gn --root=../.. -q --check gen .
ERROR at //build/config/linux/pkg_config.gni:103:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /home/ranciere/work/flutter_embedded/third_party/engine/src/out/linux_debug_arm/
Command: python /home/ranciere/work/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py -s /home/ranciere/work/sysroot -a arm --system_libdir lib gtk+-3.0
Returned 1.
stderr:

Package gtk+-3.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gtk+-3.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gtk+-3.0' found
Traceback (most recent call last):
  File "/home/ranciere/work/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py", line 248, in <module>
    sys.exit(main())
  File "/home/ranciere/work/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py", line 143, in main
    prefix = GetPkgConfigPrefixToStrip(options, args)
  File "/home/ranciere/work/flutter_embedded/third_party/engine/src/build/config/linux/pkg-config.py", line 82, in GetPkgConfigPrefixToStrip
    "--variable=prefix"] + args, env=os.environ).decode('utf-8')
  File "/usr/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['pkg-config', '--variable=prefix', 'gtk+-3.0']' returned non-zero exit status 1

See //flutter/shell/platform/linux/config/BUILD.gn:14:1: whence it was called.
pkg_config("gtk") {
^------------------
See //flutter/shell/platform/linux/BUILD.gn:67:5: which caused the file to be included.
    "//flutter/shell/platform/linux/config:gtk",
    ^------------------------------------------
ninja: error: rebuilding 'build.ninja': subcommand failed
make[2]: *** [CMakeFiles/engine.dir/build.make:112: engine-prefix/src/engine-stamp/engine-build] Error 1
make[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/engine.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

jwinarske, 2020-05-12:
When did you do a git pull last?   You may have to manually remove the
patches.  That's on my to-do list.

jwinarske, 2020-05-13:
When did you do a git pull last? You may have to manually remove the
patches. That's on my to-do list.

Also you will want to clear some CMake variables that may have been set in the CMake cache file.

`git pull`

flutter_glfw example:
`cmake .. -DBUILD_FLUTTER_RPI=ON -UTARGET_SYSROOT -UENGINE_EMBEDDER_FOR_TARGET -UENGINE_DISABLE_DESKTOP`

gtk3+ desktop shell (ignore install fail for now):
`cmake .. -DTARGET_SYSROOT=/home/ranciere/work/sysroot -UBUILD_FLUTTER_RPI -UENGINE_EMBEDDER_FOR_TARGET -DENGINE_DISABLE_DESKTOP=OFF -DENGINE_EMBEDDER_FOR_TARGET=OFF`

Check the engine build folder in the case of gtk3+ desktop shell. I would expect an executable that can be run as-is.

On a fresh clone, you can skip the -U.

jwinarske, 2020-05-13:
I worked through the gtk+-3.0 for host, default rpi config, and updated the readme.

When you specify a different TARGET_SYSROOT, you may need to override PKG_CONFIG_PATH when calling CMake.  Do this with `-DPKG_CONFIG_PATH=/mypath/to/pkgcfg:/myother/path/pkgcfg`

ranciere, 2020-05-13:
I tried in both way (```cmake .. -DBUILD_FLUTTER_RPI=ON``` and ```cmake .. -DTARGET_SYSROOT=/home/ranciere/work/sysroot -DENGINE_DISABLE_DESKTOP=OFF -DENGINE_EMBEDDER_FOR_TARGET=OFF```) in two newly cloned repository (ee6ece9) and I got this new error in both case:
```
[3806/3809] SOLINK ./libflutter_engine.so
FAILED: libflutter_engine.so so.unstripped/libflutter_engine.so libflutter_engine.so.TOC 
/home/ranciere/work/flutter_embedded_glfw/third_party/engine/src/buildtools/linux-x64/clang/bin/clang++ --target=armv7-unknown-linux-gnueabihf --sysroot /home/ranciere/work/flutter_embedded_glfw/sysroot -shared -Wl,--fatal-warnings -fPIC -Wl,-z,noexecstack -Wl,-z,now -Wl,-z,relro -Wl,-z,defs -pthread --sysroot=/home/ranciere/work/flutter_embedded_glfw/sysroot  -Wl,-O2 -Wl,--gc-sections -Wl,--as-needed -o ./so.unstripped/libflutter_engine.so  -Wl,--build-id=sha1 -Wl,-soname=libflutter_engine.so @./libflutter_engine.so.rsp && { /home/ranciere/work/flutter_embedded_glfw/third_party/engine/src/buildtools/linux-x64/clang/bin/llvm-readelf -d ./so.unstripped/libflutter_engine.so | grep SONAME ; /home/ranciere/work/flutter_embedded_glfw/third_party/engine/src/buildtools/linux-x64/clang/bin/llvm-nm -gD -f posix ./so.unstripped/libflutter_engine.so | cut -f1-2 -d' '; } > ./libflutter_engine.so.tmp && if ! cmp -s ./libflutter_engine.so.tmp ./libflutter_engine.so.TOC; then mv ./libflutter_engine.so.tmp ./libflutter_engine.so.TOC; fi && /home/ranciere/work/flutter_embedded_glfw/third_party/engine/src/buildtools/linux-x64/clang/bin/llvm-strip -o ./libflutter_engine.so ./so.unstripped/libflutter_engine.so
ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by allocatestack.c:145
>>>               pthread_create.o:(stack_list_del) in archive /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by allocatestack.c:282
>>>               pthread_create.o:(free_stacks) in archive /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by allocatestack.c:282
>>>               pthread_create.o:(free_stacks) in archive /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by allocatestack.c:160
>>>               pthread_create.o:(stack_list_add) in archive /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against symbol: start_thread in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by createthread.c:142 (../sysdeps/unix/sysv/linux/createthread.c:142)
>>>               pthread_create.o:(create_thread) in archive /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
>>> defined in /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a(pthread_create.o)
>>> referenced by createthread.c:142 (../sysdeps/unix/sysv/linux/createthread.c:142)
>>>               pthread_create.o:(create_thread) in archive /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: can't create dynamic relocation R_ARM_ABS32 against local symbol in readonly segment; recompile object files with -fPIC or pass '-Wl,-z,notext' to allow text relocations in the output
[... cut ...]
>>>               pthread_create.o:(__reclaim_stacks) in archive /home/ranciere/work/flutter_embedded_glfw/sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libpthread.a

ld.lld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
clang-11: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
make[2]: *** [CMakeFiles/engine.dir/build.make:112: engine-prefix/src/engine-stamp/engine-build] Error 1
make[1]: *** [CMakeFiles/Makefile2:140: CMakeFiles/engine.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

ranciere, 2020-05-13:
I've tried without parameters (```cmake .. -GNinja```) and the linker error exists.

jwinarske, 2020-05-13:
What's the output of your cmake invocation?

ranciere, 2020-05-13:
The output: 
```
-- CMAKE_BUILD_TYPE not set, defaulting to Release.
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Generator .............. Ninja
-- Build Type ............. Release
-- ENGINE_RUNTIME_MODE not set, defaulting to arm.
-- Rootfs Archive Url ..... http://director.downloads.raspberrypi.org/raspbian/archive/2020-02-14-13:48/root.tar.xz
-- Target Sysroot ......... /home/ranciere/work/flutter_embedded/sysroot
-- Engine Repo ............ git@github.com:flutter/engine.git
-- Engine Commit .......... 9e5072f0ce81206b99db3598da687a19ce57a863
-- ENGINE_RUNTIME_MODE not set, defaulting to debug.
-- PKG_CONFIG_PATH ........ /home/ranciere/work/flutter_embedded/sysroot/usr/lib/arm-linux-gnueabihf/pkgconfig:/usr/share/pkgconfig
-- Engine Flags ........... --runtime-mode debug --no-goma --lto --clang --enable-fontconfig --embedder-for-target --full-dart-sdk --disable-desktop-embeddings --target-os linux --linux-cpu arm --target-sysroot /home/ranciere/work/flutter_embedded/sysroot --target-toolchain /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang --target-triple armv7-unknown-linux-gnueabihf --arm-float-abi hard
-- custom_lib_flags ....... $target_sysroot/usr/lib/arm-linux-gnueabihf/libpthread.a $target_sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/libgcc_eh.a $target_sysroot/usr/lib/arm-linux-gnueabihf/libc.a -Wl,-z,notext
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ranciere/work/flutter_embedded/build
```

jwinarske, 2020-05-13:
Confirm that the values assigned to "custom_lib_flags" match your sysroot.  Override with ENGINE_CUSTOM_LIB_FLAGS.

jwinarske, 2020-05-14:
That linker error indicates the patch didn't apply.  As the patch explicitly adds "-Wl,-z,notext".  The patch step only runs once per download.  So chances are something was interrupted on the build.  You can force the patch step to run again by either editing the CMake file; as a space somewhere, or deleting one of the files under engine-prefix.

jwinarske, 2020-05-15:
Okay.  I hit the same problem, and I think I have a fix.  It's in the build patch. Although it's odd as I've gotten past this with the old patch, so I'm not convinced it's totally resolved.

Three options after pulling:
`rm .../build/engine-prefix/src/engine-stamp/engine-done`
`rm .../build/engine-prefix/src/engine-stamp/engine-patch`
make

this forces patch to happen again

- or -
blow away build/engine-prefix *and* third_party/engine

- or -
cd third_party/engine/src
git checkout .
cd third_party/icu
git checkout .

If you hit the pkg_config issue on a re-build, you can get around that by exporting PKG_CONFIG_PATH pointing to the right pkgconfig folder.  In the case of the default build it points to: `${TARGET_SYSROOT}/usr/lib/arm-linux-gnueabihf/pkgconfig:/usr/share/pkgconfig`

ranciere, 2020-05-15:
Hi,
Unfortunately no progress, the bug still exists wit a fresh checkout (rm -fr ./flutter_embeded) than I followed the instructions (git clone; export PKG_CONFIG_PATH; cmake ..; make). The cmake config:
```
-- CMAKE_BUILD_TYPE not set, defaulting to Release.
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Generator .............. Unix Makefiles
-- Build Type ............. Release
-- ENGINE_RUNTIME_MODE not set, defaulting to arm.
-- Rootfs Archive Url ..... http://director.downloads.raspberrypi.org/raspbian/archive/2020-02-14-13:48/root.tar.xz
-- Target Sysroot ......... /home/ranciere/work/flutter_embedded/sysroot
-- Engine Repo ............ git@github.com:flutter/engine.git
-- Engine Commit .......... 9e5072f0ce81206b99db3598da687a19ce57a863
-- ENGINE_RUNTIME_MODE not set, defaulting to debug.
-- PKG_CONFIG_PATH ........ /home/ranciere/work/flutter_embedded/sysroot/usr/lib/arm-linux-gnueabihf/pkgconfig:/usr/share/pkgconfig
-- Engine Flags ........... --runtime-mode debug --no-goma --lto --clang --enable-fontconfig --embedder-for-target --full-dart-sdk --disable-desktop-embeddings --target-os linux --linux-cpu arm --target-sysroot /home/ranciere/work/flutter_embedded/sysroot --target-toolchain /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang --target-triple armv7-unknown-linux-gnueabihf --arm-float-abi hard
-- custom_lib_flags ....... $target_sysroot/usr/lib/arm-linux-gnueabihf/libpthread.a $target_sysroot/usr/lib/gcc/arm-linux-gnueabihf/8/libgcc_eh.a $target_sysroot/usr/lib/arm-linux-gnueabihf/libc.a -Wl,-z,notext
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ranciere/work/flutter_embedded/build
```
The PKG_CONFIG_PATH is correct, the pc files are presented there. How can I check a build patches? Could you tell me a relative path to a makefile and build flags to look at?

jwinarske, 2020-05-16:
You should be good to go now.  I've added Channel support, and have checked all the channels for armv7.  Once you confirm you're good I'm going to mark a release.

ranciere, 2020-05-17:
Hi @jwinarske,
Interesting sequel, I have another bug during compilation of the engine (new checkout):
```
[1973/3804] AR obj/third_party/dart/runtime/libdart_jit.a
FAILED: obj/third_party/dart/runtime/libdart_jit.a 
rm -f obj/third_party/dart/runtime/libdart_jit.a && /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/armv7-unknown-linux-gnueabihf-ar rcs obj/third_party/dart/runtime/libdart_jit.a @obj/third_party/dart/runtime/libdart_jit.a.rsp
/bin/sh: 1: /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/armv7-unknown-linux-gnueabihf-ar: not found
[1996/3804] CXX obj/third_party/libcxx/src/libcxx.locale.o
ninja: build stopped: subcommand failed.
```
But if I rerun the make -j then the compilation will be successful :o

ranciere, 2020-05-17:
I've installed the executable to /usr/bin/flutter_glfw and the dynamic library to /usr/lib/arm-linux-gnueabihf/libflutter_engine.so; but the executable cannot load the flutter lib:
`flutter_glfw: error while loading shared libraries: libflutter_engine.so: cannot open shared object file: No such file or directory
`
If I run it via strace the executable is able to open the lib:
```
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\0\240\10\0014\0\0\0"..., 512) = 512
_llseek(3, 25935456, [25935456], SEEK_SET) = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1720) = 1720
close(3)                                = 0
```
I think the linker flags/architecture do not fit. Should I open a new ticket for this?

jwinarske, 2020-05-17:
what's the output of this: readelf -d libflutter_engine.so

jwinarske, 2020-05-17:
also the icu dat file is required in the same directory as the engine so

ranciere, 2020-05-17:
Readelf output:
```
Dynamic section at offset 0x18b9778 contains 25 entries:
  Tag        Type                         Name/Value
 0x0000000e (SONAME)                     Library soname: [libflutter_engine.so]
 0x0000001e (FLAGS)                      TEXTREL BIND_NOW
 0x6ffffffb (FLAGS_1)                    Flags: NOW
 0x00000011 (REL)                        0x21534
 0x00000012 (RELSZ)                      574704 (bytes)
 0x00000013 (RELENT)                     8 (bytes)
 0x6ffffffa (RELCOUNT)                   68923
 0x00000017 (JMPREL)                     0xada24
 0x00000002 (PLTRELSZ)                   5640 (bytes)
 0x00000003 (PLTGOT)                     0x18baca8
 0x00000014 (PLTREL)                     REL
 0x00000006 (SYMTAB)                     0xe24
 0x0000000b (SYMENT)                     16 (bytes)
 0x00000005 (STRTAB)                     0x144a0
 0x0000000a (STRSZ)                      53395 (bytes)
 0x00000016 (TEXTREL)                    0x0
 0x6ffffef5 (GNU_HASH)                   0xb0f4
 0x00000004 (HASH)                       0xf330
 0x00000019 (INIT_ARRAY)                 0x186f01c
 0x0000001b (INIT_ARRAYSZ)               692 (bytes)
 0x0000001a (FINI_ARRAY)                 0x186f014
 0x0000001c (FINI_ARRAYSZ)               8 (bytes)
 0x0000000c (INIT)                       0x18675d0
 0x0000000d (FINI)                       0x18675dc
 0x00000000 (NULL)                       0x0
```
I've put the icudtl.dat to the /usr/lib/arm-linux-gnueabihf directory, but results is the same.

jwinarske, 2020-05-17:
now run readelf -d on the executable

ranciere, 2020-05-17:
Here is:

```
Dynamic section at offset 0x105fa0 contains 29 entries:
  Tag        Type                         Name/Value
 0x00000001 (NEEDED)                     Shared library: [libflutter_engine.so]
 0x00000001 (NEEDED)                     Shared library: [libc.so.6]
 0x00000001 (NEEDED)                     Shared library: [ld-linux-armhf.so.3]
 0x0000001e (FLAGS)                      TEXTREL
 0x00000015 (DEBUG)                      0x0
 0x00000011 (REL)                        0x1d4b4
 0x00000012 (RELSZ)                      1696 (bytes)
 0x00000013 (RELENT)                     8 (bytes)
 0x00000017 (JMPREL)                     0x1db54
 0x00000002 (PLTRELSZ)                   696 (bytes)
 0x00000003 (PLTGOT)                     0x113500
 0x00000014 (PLTREL)                     REL
 0x00000006 (SYMTAB)                     0x126a0
 0x0000000b (SYMENT)                     16 (bytes)
 0x00000005 (STRTAB)                     0x19980
 0x0000000a (STRSZ)                      15154 (bytes)
 0x00000016 (TEXTREL)                    0x0
 0x6ffffef5 (GNU_HASH)                   0x16608
 0x00000004 (HASH)                       0x17d58
 0x00000019 (INIT_ARRAY)                 0x114018
 0x0000001b (INIT_ARRAYSZ)               8 (bytes)
 0x0000001a (FINI_ARRAY)                 0x114010
 0x0000001c (FINI_ARRAYSZ)               8 (bytes)
 0x0000000c (INIT)                       0x11051c
 0x0000000d (FINI)                       0x110528
 0x6ffffff0 (VERSYM)                     0x15ee0
 0x6ffffffe (VERNEED)                    0x165e8
 0x6fffffff (VERNEEDNUM)                 1
 0x00000000 (NULL)                       0x0
```

ranciere, 2020-05-17:
And output of `ldd flutter_glfw` is:
```
	linux-vdso.so.1 (0xbef90000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb6f2d000)
	libflutter_engine.so => not found
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6dc7000)
	/lib/ld-linux-armhf.so.3 (0xb6f42000)
```

jwinarske, 2020-05-17:
This is what I see:

```
readelf -d ./flutter_glfw 

Dynamic section at offset 0x104fa0 contains 29 entries:
  Tag        Type                         Name/Value
 0x0000001d (RUNPATH)                    Library runpath: [/media/joel/SolidState/flutter_embedded/build/stable:]
 0x00000001 (NEEDED)                     Shared library: [libflutter_engine.so]
 0x00000001 (NEEDED)                     Shared library: [libc.so.6]
 0x00000001 (NEEDED)                     Shared library: [ld-linux-armhf.so.3]

```

Running the executable it attempts to resolve libraries based on the relative path.  If your target doesn't have this path it will fail to find the dependencies.

To get it to run you need to do this:
LD_LIBRARY_PATH=/usr/lib/arm-linux-gnueabihf flutter_glfw

The RPATH isn't a problem in the case of a Yocto build, as Yocto implements a virtual sysroot, and it installs to the path it will run from.

jwinarske, 2020-05-17:
The flutter example CMake did not have an install step.  Without this it creates the bogus rpath, which is what you're looking at.  I fixed it last night.  Make sure you have that fix `git pull`

From master
```
readelf -d /media/joel/SolidState/flutter_embedded/sysroot/usr/bin/flutter_glfw

Dynamic section at offset 0x104fa0 contains 28 entries:
  Tag        Type                         Name/Value
 0x00000001 (NEEDED)                     Shared library: [libflutter_engine.so]
 0x00000001 (NEEDED)                     Shared library: [libc.so.6]
 0x00000001 (NEEDED)                     Shared library: [ld-linux-armhf.so.3]

```

ranciere, 2020-05-18:
At first I thought that LD_LIBRARY_PATH was the problem, but 
Here is a the strace log of loading of libflutter_engine.so without so:
```
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/tls/v7l/neon/vfp/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/tls/v7l/neon/vfp", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/tls/v7l/neon/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/tls/v7l/neon", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/tls/v7l/vfp/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/tls/v7l/vfp", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/tls/v7l/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/tls/v7l", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/tls/neon/vfp/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/tls/neon/vfp", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/tls/neon/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/tls/neon", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/tls/vfp/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/tls/vfp", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/tls/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/tls", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/v7l/neon/vfp/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/v7l/neon/vfp", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/v7l/neon/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/v7l/neon", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/v7l/vfp/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/v7l/vfp", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/v7l/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/v7l", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/neon/vfp/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/neon/vfp", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/neon/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/neon", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/vfp/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf/vfp", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/lib/arm-linux-gnueabihf", {st_mode=S_IFDIR|0755, st_size=12288, ...}) = 0
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/tls/v7l/neon/vfp/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/tls/v7l/neon/vfp", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/tls/v7l/neon/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/tls/v7l/neon", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/tls/v7l/vfp/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/tls/v7l/vfp", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/tls/v7l/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/tls/v7l", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/tls/neon/vfp/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/tls/neon/vfp", 0xbe92f9c8) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/tls/neon/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[... cut ...] ollowings (part of it):
```
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\0\240\10\0014\0\0\0"..., 512) = 512
_llseek(3, 25935456, [25935456], SEEK_SET) = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1720) = 1720
close(3)
```
As you see, the system is able to open and read the library, so this is not the fault. 
(I've defined the TARGET_SYSROOT).

jwinarske, 2020-05-18:
Is there an rpath on your library?  If so, this is from an outdated build.

I just tested package installing, after both packages installed, using strace I see it load the library, but fails for another reason.  More fun later :)

jwinarske, 2020-05-19:
Let's open a different issue to match what we're looking at.  I think the build finds the missing library now.  Do you agree?

jwinarske, 2020-05-19:
closing since we are no longer `missing flutter_linux library`.

closed 2020-05-19 by jwinarske

ranciere, 2020-05-19:
Yes, I agree! I will open a new issue today!