meta-flutter/flutter_embedded #35: flutter_glfw cannot link

ranciere, 2020-05-23:
config is default(`cmake ..`). Error message:
```
[100%] Linking CXX executable flutter_glfw
ld.lld: error: undefined symbol: pthread_mutex_trylock
>>> referenced by memory.cpp
>>>               memory.cpp.o:(std::__2::__sp_mut::lock()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_mutex_trylock
>>> referenced by memory.cpp
>>>               memory.cpp.o:(std::__2::__sp_mut::lock()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_mutex_trylock
>>> referenced by mutex.cpp
>>>               mutex.cpp.o:(std::__2::mutex::try_lock()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_mutexattr_init
>>> referenced by mutex.cpp
>>>               mutex.cpp.o:(std::__2::recursive_mutex::recursive_mutex()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_mutexattr_settype
>>> referenced by mutex.cpp
>>>               mutex.cpp.o:(std::__2::recursive_mutex::recursive_mutex()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_mutexattr_destroy
>>> referenced by mutex.cpp
>>>               mutex.cpp.o:(std::__2::recursive_mutex::recursive_mutex()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_mutexattr_destroy
>>> referenced by mutex.cpp
>>>               mutex.cpp.o:(std::__2::recursive_mutex::recursive_mutex()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_mutex_trylock
>>> referenced by mutex.cpp
>>>               mutex.cpp.o:(std::__2::recursive_mutex::try_lock()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_mutex_trylock
>>> referenced by mutex.cpp
>>>               mutex.cpp.o:(std::__2::timed_mutex::try_lock()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_mutex_trylock
>>> referenced by mutex.cpp
>>>               mutex.cpp.o:(std::__2::recursive_timed_mutex::try_lock()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_getspecific
>>> referenced by condition_variable.cpp
>>>               condition_variable.cpp.o:(std::__2::notify_all_at_thread_exit(std::__2::condition_variable&, std::__2::unique_lock<std::__2::mutex>)) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_setspecific
>>> referenced by condition_variable.cpp
>>>               condition_variable.cpp.o:(std::__2::notify_all_at_thread_exit(std::__2::condition_variable&, std::__2::unique_lock<std::__2::mutex>)) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_getspecific
>>> referenced by condition_variable.cpp
>>>               condition_variable.cpp.o:(std::__2::notify_all_at_thread_exit(std::__2::condition_variable&, std::__2::unique_lock<std::__2::mutex>)) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_join
>>> referenced by thread.cpp
>>>               thread.cpp.o:(std::__2::thread::join()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_detach
>>> referenced by thread.cpp
>>>               thread.cpp.o:(std::__2::thread::detach()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_key_create
>>> referenced by thread.cpp
>>>               thread.cpp.o:(std::__2::__thread_local_data()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_getspecific
>>> referenced by future.cpp
>>>               future.cpp.o:(std::__2::__assoc_sub_state::set_value_at_thread_exit()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_getspecific
>>> referenced by future.cpp
>>>               future.cpp.o:(std::__2::__assoc_sub_state::set_exception_at_thread_exit(std::exception_ptr)) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [CMakeFiles/flutter_glfw.dir/build.make:86: flutter_glfw] Error 1
make[4]: *** [CMakeFiles/Makefile2:76: CMakeFiles/flutter_glfw.dir/all] Error 2
make[3]: *** [Makefile:152: all] Error 2
make[2]: *** [CMakeFiles/glfw_flutter.dir/build.make:113: glfw_flutter-prefix/src/glfw_flutter-stamp/glfw_flutter-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:196: CMakeFiles/glfw_flutter.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

jwinarske, 2020-05-24:
I'll take a look this coming week, out on hoiidays

ranciere, 2020-05-27:
Additional information:
If I add the pthread dependency ```
find_package(Threads REQUIRED)
target_link_libraries(flutter_glfw ${CMAKE_THREAD_LIBS_INIT})
)
``` to flutter_glfw (thirdparty/engine/src/flutter/examples/glfw/CMakeLists.txt) then the MCake configure step will fail. The CMakeError.log:
Performing C++ SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
```
Change Dir: /home/ranciere/work/flutter_embedded/buil/glfw_flutter-prefix/src/glfw_flutter-build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_baa0c/fast && make[4]: Entering directory '/home/ranciere/work/flutter_embedded/buil/glfw_flutter-prefix/src/glfw_flutter-build/CMakeFiles/CMakeTmp'
/usr/bin/make -f CMakeFiles/cmTC_baa0c.dir/build.make CMakeFiles/cmTC_baa0c.dir/build
make[5]: Entering directory '/home/ranciere/work/flutter_embedded/buil/glfw_flutter-prefix/src/glfw_flutter-build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_baa0c.dir/src.cxx.o
/home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/clang++ --target=armv7-neon-vfpv4-linux-gnueabihf --sysroot=/home/ranciere/work/flutter_embedded/sysroot    -I/home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/include -I/home/ranciere/work/flutter_embedded/sysroot/opt/vc/include -DGLFW_EXPOSE_NATIVE_EGL -DGLFW_INCLUDE_ES2 -flto=thin -std=c++11 -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_baa0c.dir/src.cxx.o -c /home/ranciere/work/flutter_embedded/buil/glfw_flutter-prefix/src/glfw_flutter-build/CMakeFiles/CMakeTmp/src.cxx
Linking CXX executable cmTC_baa0c
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_baa0c.dir/link.txt --verbose=1
/home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/clang++ --target=armv7-neon-vfpv4-linux-gnueabihf --sysroot=/home/ranciere/work/flutter_embedded/sysroot  -I/home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/include -I/home/ranciere/work/flutter_embedded/sysroot/opt/vc/include -DGLFW_EXPOSE_NATIVE_EGL -DGLFW_INCLUDE_ES2 -flto=thin -std=c++11 -DCMAKE_HAVE_LIBC_PTHREAD  -Wl,-rpath,'$ORIGIN/usr/lib/arm-linux-gnueabihf/' -L/home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib -fuse-ld=lld -L/home/ranciere/work/flutter_embedded/sysroot/lib/arm-linux-gnueabihf  CMakeFiles/cmTC_baa0c.dir/src.cxx.o  -o cmTC_baa0c 
ld.lld: error: undefined symbol: dladdr
>>> referenced by libunwind.cpp
>>>               libunwind.cpp.o:(libunwind::UnwindCursor<libunwind::LocalAddressSpace, libunwind::Registers_arm>::getFunctionName(char*, unsigned int, unsigned int*)) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_once
>>> referenced by cxa_exception_storage.cpp
>>>               cxa_exception_storage.cpp.o:(__cxa_get_globals) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_getspecific
>>> referenced by cxa_exception_storage.cpp
>>>               cxa_exception_storage.cpp.o:(__cxa_get_globals) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_setspecific
>>> referenced by cxa_exception_storage.cpp
>>>               cxa_exception_storage.cpp.o:(__cxa_get_globals) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_once
>>> referenced by cxa_exception_storage.cpp
>>>               cxa_exception_storage.cpp.o:(__cxa_get_globals_fast) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_getspecific
>>> referenced by cxa_exception_storage.cpp
>>>               cxa_exception_storage.cpp.o:(__cxa_get_globals_fast) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_key_create
>>> referenced by cxa_exception_storage.cpp
>>>               cxa_exception_storage.cpp.o:(__cxxabiv1::(anonymous namespace)::construct_()) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_setspecific
>>> referenced by cxa_exception_storage.cpp
>>>               cxa_exception_storage.cpp.o:(__cxxabiv1::(anonymous namespace)::destruct_(void*)) in archive /home/ranciere/work/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/lib/clang/8.0.0/armv7-linux-gnueabihf/lib/libc++.a

ld.lld: error: undefined symbol: pthread_create
>>> referenced by src.cxx
>>>               lto.tmp:(main)

ld.lld: error: undefined symbol: pthread_detach
>>> referenced by src.cxx
>>>               lto.tmp:(main)

ld.lld: error: undefined symbol: pthread_join
>>> referenced by src.cxx
>>>               lto.tmp:(main)
[... cut ...] abihf/lib/libc++.a
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [CMakeFiles/cmTC_54c9a.dir/build.make:87: cmTC_54c9a] Error 1
make[5]: Leaving directory '/home/ranciere/work/flutter_embedded/buil/glfw_flutter-prefix/src/glfw_flutter-build/CMakeFiles/CMakeTmp'
make[4]: *** [Makefile:121: cmTC_54c9a/fast] Error 2
make[4]: Leaving directory '/home/ranciere/work/flutter_embedded/buil/glfw_flutter-prefix/src/glfw_flutter-build/CMakeFiles/CMakeTmp'
```

jwinarske, 2020-05-28:
Without changing anything, what do you get when you do this?

    make symlink_fixups
    make

jwinarske, 2020-05-28:
This should be resolved in master.  Two fixes:
1. make engine depend on symlink_fixups target instead of sysroot.  This was causing jit build break.
2. Add -lpthread link flag for flutter_glfw

ranciere, 2020-05-28:
Great! I've checked, it is working. I close the ticket.

closed 2020-05-28 by ranciere