meta-flutter/flutter_embedded #36: libflutter so cannot be loaded

ranciere, 2020-05-29:
The library loading problem is still an issue. Output of running flutter_glfw:
`/flutter_glfw: error while loading shared libraries: libflutter_engine.so: cannot open shared object file: No such file or directory`
The libflutter_engine.so is in `usr/lib/arm-linux-gnueabihf/libflutter_engine.so`. The executable tries to load it, but it does not successful.

jwinarske, 2020-05-29:
I'll take a look this weekend

ranciere, 2020-06-08:
Hi @jwinarske Is there any progress on the issue?

jwinarske, 2020-06-10:
@ranciere not yet.  Just finished setting up my new build machine.  Soon

ranciere, 2020-06-10:
I've also bought a new machine (Ryzen 9 3900X) because of the compilation time :D

ranciere, 2020-06-20:
Hi @jwinarske Could you deal with it?

jwinarske, 2020-06-20:
I'll take a look later today as part of my vulkan driver integration tests.

jwinarske, 2020-07-01:
```
pi@raspberrypi:/lib/arm-linux-gnueabihf $ ldd -r -v ./libflutter_engine.so 
	not a dynamic executable
```

It should look something like this
```
pi@raspberrypi:/lib/arm-linux-gnueabihf $ ldd -v libm.so.6 
	linux-vdso.so.1 (0x7ec33000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76ea6000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76d42000)
	/lib/ld-linux-armhf.so.3 (0x76f3d000)

	Version information:
	./libm.so.6:
		ld-linux-armhf.so.3 (GLIBC_2.4) => /lib/ld-linux-armhf.so.3
		libc.so.6 (GLIBC_PRIVATE) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libc.so.6
	/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so:
		libc.so.6 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libc.so.6
	/lib/arm-linux-gnueabihf/libc.so.6:
		ld-linux-armhf.so.3 (GLIBC_2.4) => /lib/ld-linux-armhf.so.3
		ld-linux-armhf.so.3 (GLIBC_PRIVATE) => /lib/ld-linux-armhf.so.3
```

jwinarske, 2020-07-01:
What do you get from `strace ldd -v /usr/lib/arm-linux-gnueabihf/libflutter_engine.so`?

ranciere, 2020-07-10:
Hi @jwinarske ,
ldd output:
```
pi@raspberrypi:~ $ ldd -r -v /usr/lib/arm-linux-gnueabihf/libflutter_engine.so 
	not a dynamic executable
```
The output of the stracing:
```
pi@raspberrypi:~ $ strace ldd -v /usr/lib/arm-linux-gnueabihf/libflutter_engine.so
execve("/usr/bin/ldd", ["ldd", "-v", "/usr/lib/arm-linux-gnueabihf/lib"...], 0xbeede568 /* 34 vars */) = 0
brk(NULL)                               = 0x77b000
uname({sysname="Linux", nodename="raspberrypi", ...}) = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f26000
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0xb6f25000
close(3)                                = 0
readlink("/proc/self/exe", "/bin/bash", 4096) = 9
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\3\0\0004\0\0\0"..., 512) = 512
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
fstat64(3, {st_mode=S_IFREG|0644, st_size=17708, ...}) = 0
mmap2(NULL, 81964, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ee3000
mprotect(0xb6ee7000, 61440, PROT_NONE)  = 0
mmap2(0xb6ef6000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0xb6ef6000
close(3)                                = 0
munmap(0xb6f25000, 54)                  = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=97821, ...}) = 0
mmap2(NULL, 97821, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6ecb000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libtinfo.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0pn\0\0004\0\0\0"..., 512) = 512
_llseek(3, 132176, [132176], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 131828, [131828], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 132176, [132176], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 131828, [131828], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 132176, [132176], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 131828, [131828], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 132176, [132176], SEEK_SET)  = 0
[... 232 lines cut ...]
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, [INT CHLD], [], 8) = 0
clone(child_stack=NULL, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0xb6f24078) = 20022
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
rt_sigaction(SIGCHLD, {sa_handler=0x55a6c, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6d66120}, {sa_handler=0x55a6c, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6d66120}, 8) = 0
close(4)                                = 0
read(3, "", 128)                        = 0
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=20022, si_uid=1000, si_status=1, si_utime=0, si_stime=1} ---
wait4(-1, [{WIFEXITED(s) && WEXITSTATUS(s) == 1}], WNOHANG, NULL) = 20022
wait4(-1, 0xbed79054, WNOHANG, NULL)    = -1 ECHILD (No child processes)
sigreturn({mask=[]})                    = 0
close(3)                                = 0
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigaction(SIGINT, {sa_handler=0x526b4, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6d66120}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6d66120}, 8) = 0
[... cut ...]  {sa_handler=0x526b4, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6d66120}, 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
fstat64(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0), ...}) = 0
write(1, "\tnot a dynamic executable\n", 26	not a dynamic executable
) = 26
read(255, "\nexit $result\n# Local Variables:"..., 5313) = 52
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
exit_group(1)                           = ?
+++ exited with 1 +++
```

jwinarske, 2020-07-14:
It looks like it fails to load locale resources, then fails.  Which is why I tagged this "locale".

```
openat(AT_FDCWD, "/usr/share/locale/en_US.UTF-8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en_US.utf8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en_US/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en.UTF-8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en.utf8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
stat64("/usr/lib/arm-linux-gnueabihf/libflutter_engine.so", {st_mode=S_IFREG|0644, st_size=27140084, ...}) = 0
stat64("/usr/lib/arm-linux-gnueabihf/libflutter_engine.so", {st_mode=S_IFREG|0644, st_size=27140084, ...}) = 0
faccessat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libflutter_engine.so", R_OK) = 0
```

DisDis, 2020-07-26:
i have same problem.
ldd -v -r ./libflutter_engine.so 

> 	not a dynamic executable

readelf -d ./libflutter_engine.so | grep 'NEEDED'

>  0x00000001 (NEEDED)                     Shared library: [libdl.so.2]
>  0x00000001 (NEEDED)                     Shared library: [libpthread.so.0]
>  0x00000001 (NEEDED)                     Shared library: [librt.so.1]
>  0x00000001 (NEEDED)                     Shared library: [libm.so.6]
>  0x00000001 (NEEDED)                     Shared library: [libc.so.6]
>  0x00000001 (NEEDED)                     Shared library: [ld-linux-armhf.so.3]

If i take https://github.com/ardera/flutter-pi/blob/engine-binaries/libflutter_engine.so and run readelf -d ./libflutter_engine.so | grep 'NEEDED'

>  0x00000001 (NEEDED)                     Shared library: [libdl.so.2]
>  0x00000001 (NEEDED)                     Shared library: [libpthread.so.0]
>  0x00000001 (NEEDED)                     Shared library: [librt.so.1]
>  0x00000001 (NEEDED)                     Shared library: [libm.so.6]
>  0x00000001 (NEEDED)                     Shared library: [libgcc_s.so.1]
>  0x00000001 (NEEDED)                     Shared library: [libc.so.6]
>  0x00000001 (NEEDED)                     Shared library: [ld-linux-armhf.so.3]

and
```
	linux-vdso.so.1 (0xbed76000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb577b000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb5753000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb5729000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb5712000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb5690000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb5663000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb5515000)
	/lib/ld-linux-armhf.so.3 (0xb6ed0000)

	Version information:
	./libflutter_engine.so:
		ld-linux-armhf.so.3 (GLIBC_2.4) => /lib/ld-linux-armhf.so.3
		librt.so.1 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/librt.so.1
		libdl.so.2 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libdl.so.2
		libgcc_s.so.1 (GCC_3.0) => /lib/arm-linux-gnueabihf/libgcc_s.so.1
		libgcc_s.so.1 (GCC_3.5) => /lib/arm-linux-gnueabihf/libgcc_s.so.1
		libpthread.so.0 (GLIBC_2.12) => /lib/arm-linux-gnueabihf/libpthread.so.0
		libpthread.so.0 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libpthread.so.0
		libm.so.6 (GLIBC_2.27) => /lib/arm-linux-gnueabihf/libm.so.6
		libm.so.6 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libm.so.6
		libc.so.6 (GLIBC_2.8) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_2.9) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_2.6) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_2.16) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_2.28) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_2.7) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libc.so.6
	/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so:
		libc.so.6 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libc.so.6
	/lib/arm-linux-gnueabihf/libdl.so.2:
		ld-linux-armhf.so.3 (GLIBC_PRIVATE) => /lib/ld-linux-armhf.so.3
		ld-linux-armhf.so.3 (GLIBC_2.4) => /lib/ld-linux-armhf.so.3
		libc.so.6 (GLIBC_PRIVATE) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libc.so.6
	/lib/arm-linux-gnueabihf/libpthread.so.0:
		ld-linux-armhf.so.3 (GLIBC_PRIVATE) => /lib/ld-linux-armhf.so.3
		ld-linux-armhf.so.3 (GLIBC_2.4) => /lib/ld-linux-armhf.so.3
		libc.so.6 (GLIBC_PRIVATE) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libc.so.6
	/lib/arm-linux-gnueabihf/librt.so.1:
		ld-linux-armhf.so.3 (GLIBC_PRIVATE) => /lib/ld-linux-armhf.so.3
		ld-linux-armhf.so.3 (GLIBC_2.4) => /lib/ld-linux-armhf.so.3
		libc.so.6 (GLIBC_2.28) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_PRIVATE) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libc.so.6
		libpthread.so.0 (GLIBC_PRIVATE) => /lib/arm-linux-gnueabihf/libpthread.so.0
		libpthread.so.0 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libpthread.so.0
	/lib/arm-linux-gnueabihf/libm.so.6:
		ld-linux-armhf.so.3 (GLIBC_2.4) => /lib/ld-linux-armhf.so.3
		libc.so.6 (GLIBC_PRIVATE) => /lib/arm-linux-gnueabihf/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libc.so.6
	/lib/arm-linux-gnueabihf/libgcc_s.so.1:
		libc.so.6 (GLIBC_2.4) => /lib/arm-linux-gnueabihf/libc.so.6
	/lib/arm-linux-gnueabihf/libc.so.6:
		ld-linux-armhf.so.3 (GLIBC_2.4) => /lib/ld-linux-armhf.so.3
		ld-linux-armhf.so.3 (GLIBC_PRIVATE) => /lib/ld-linux-armhf.so.3

```

DisDis, 2020-07-27:
cmake .. -DCHANNEL=stable -DENGINE_RUNTIME_MODE=release
```
-- CMAKE_BUILD_TYPE not set, defaulting to Release.
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Generator .............. Unix Makefiles
-- Build Type ............. Release
-- TARGET_ARCH not set, defaulting to arm
-- TARGET_SYSROOT not set, defaulting to /home/dis/workspace/flutter/flutter_embedded/sysroot
-- Rootfs Archive Url ..... http://director.downloads.raspberrypi.org/raspbian/archive/2020-02-14-13:48/root.tar.xz
-- Target Sysroot ......... /home/dis/workspace/flutter/flutter_embedded/sysroot
-- Engine Repo ............ git@github.com:flutter/engine.git
-- Flutter Channel ........ stable
-- Engine SHA1 ............ ee76268252c22f5c11e82a7b87423ca3982e51a7
-- ENGINE_LIB_FLAGS........ -Wl,-z,notext -nostdlib++ -fuse-ld=lld
-- PKG_CONFIG_PATH ........ /home/dis/workspace/flutter/flutter_embedded/sysroot/usr/lib/arm-linux-gnueabihf/pkgconfig:/usr/share/pkgconfig
-- Engine Flags ........... --runtime-mode release --no-goma --lto --clang --enable-fontconfig --embedder-for-target --full-dart-sdk --target-os linux --linux-cpu arm --target-sysroot /home/dis/workspace/flutter/flutter_embedded/sysroot --target-toolchain /home/dis/workspace/flutter/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang --target-triple armv7-neon-vfpv4-linux-gnueabihf --arm-float-abi hard
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dis/workspace/flutter/flutter_embedded/build

```

DisDis, 2020-07-27:
@jwinarske i think it is not locale problem.
because 
If i take https://github.com/ardera/flutter-pi/blob/engine-binaries/libflutter_engine.so 
> strace ldd -v -r ./libflutter_engine.so 

```
execve("/usr/bin/ldd", ["ldd", "-v", "-r", "./libflutter_engine.so"], 0xbeaf566c /* 23 vars */) = 0
brk(NULL)                               = 0x1e4f000
uname({sysname="Linux", nodename="raspberrypi", ...}) = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6fde000
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0xb6fdd000
close(3)                                = 0
readlink("/proc/self/exe", "/bin/bash", 4096) = 9
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\3\0\0004\0\0\0"..., 512) = 512
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
_llseek(3, 16748, [16748], SEEK_SET)    = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 16424, [16424], SEEK_SET)    = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\2\n\3\f\1\22\4\24"..., 45) = 45
fstat64(3, {st_mode=S_IFREG|0644, st_size=17708, ...}) = 0
mmap2(NULL, 81964, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f9b000
mprotect(0xb6f9f000, 61440, PROT_NONE)  = 0
mmap2(0xb6fae000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0xb6fae000
close(3)                                = 0
munmap(0xb6fdd000, 54)                  = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=84720, ...}) = 0
mmap2(NULL, 84720, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f86000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libtinfo.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0pn\0\0004\0\0\0"..., 512) = 512
_llseek(3, 132176, [132176], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 131828, [131828], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 132176, [132176], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 131828, [131828], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 132176, [132176], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 131828, [131828], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 132176, [132176], SEEK_SET)  = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 131828, [131828], SEEK_SET)  = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
[... 278 lines cut ...]
access("/lib/terminfo/x/xterm-256color", R_OK) = 0
openat(AT_FDCWD, "/lib/terminfo/x/xterm-256color", O_RDONLY|O_LARGEFILE) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=3535, ...}) = 0
read(3, "\36\2%\0&\0\17\0\235\1\356\5xterm-256color|xterm"..., 32768) = 3535
read(3, "", 28672)                      = 0
close(3)                                = 0
gettimeofday({tv_sec=1595809376, tv_usec=34356}, NULL) = 0
ioctl(1, TCGETS, {B38400 opost isig icanon echo ...}) = 0
ioctl(1, TCGETS, {B38400 opost isig icanon echo ...}) = 0
ioctl(1, TCGETS, {B38400 opost isig icanon echo ...}) = 0
ioctl(1, TCGETS, {B38400 opost isig icanon echo ...}) = 0
ioctl(1, TIOCGWINSZ, {ws_row=48, ws_col=195, ws_xpixel=0, ws_ypixel=0}) = 0
ioctl(0, TIOCGWINSZ, {ws_row=48, ws_col=195, ws_xpixel=0, ws_ypixel=0}) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=2662, si_uid=1000, si_status=0, si_utime=0, si_stime=1} ---
wait4(-1, 0xbeac15bc, WNOHANG, NULL)    = -1 ECHILD (No child processes)
sigreturn({mask=[]})                    = 0
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
read(255, "\nexit $result\n# Local Variables:"..., 5313) = 52
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
exit_group(0)                           = ?
+++ exited with 0 +++
```

DisDis, 2020-07-27:
Maybe this is our problem?
https://stackoverflow.com/questions/6150000/cross-compiler-ldd

DisDis, 2020-07-27:
I solved problem. @jwinarske @ranciere 
If you change Flutter Channel to 'beta', it works
-- Flutter Channel ........ beta
-- Engine SHA1 ............ 60b269d898cbe0be27e9b9ba9d21eae97b887ab6
flutter-engine-beta $ ldd ./libflutter_engine.so 
```
	linux-vdso.so.1 (0xbee53000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb54e1000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb54b9000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb548f000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb540d000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb52bf000)
	/lib/ld-linux-armhf.so.3 (0xb6f29000)
```

jwinarske, 2020-07-28:
@DisDis that's called a workaround, not a solution ;)

DisDis, 2020-07-28:
@jwinarske but new flutter will release after ~1 month, and beta will be stable:)

DisDis, 2020-07-28:
Yes, problem in clang 8.0, but how can we solve it?

jwinarske, 2020-07-28:
beta has clang 11, prior has clang 8.
Workaround for now:  https://github.com/jwinarske/flutter_embedded/pull/42

DisDis, 2020-07-30:
I found main issue.
flutter-engine-release $ strace /lib/ld-linux-armhf.so.3 --verify ./libflutter_engine.so 
**NOT WORK**
```
execve("/lib/ld-linux-armhf.so.3", ["/lib/ld-linux-armhf.so.3", "--verify", "./libflutter_engine.so"], 0xbeb7b648 /* 23 vars */) = 0
brk(NULL)                               = 0x119f000
openat(AT_FDCWD, "./libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\0\320\23\0004\0\0\0"..., 512) = 512
_llseek(3, 9103084, [9103084], SEEK_SET) = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1160) = 1160
close(3)                                = 0
exit_group(1)                           = ?
+++ exited with 1 +++
```

flutter-engine-beta-release $ strace /lib/ld-linux-armhf.so.3 --verify ./libflutter_engine.so 
**WORK**
```
execve("/lib/ld-linux-armhf.so.3", ["/lib/ld-linux-armhf.so.3", "--verify", "./libflutter_engine.so"], 0xbe9b4648 /* 23 vars */) = 0
brk(NULL)                               = 0x1417000
openat(AT_FDCWD, "./libflutter_engine.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\2002\24\0004\0\0\0"..., 512) = 512
_llseek(3, 9161672, [9161672], SEEK_SET) = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1160) = 1160
_llseek(3, 9161364, [9161364], SEEK_SET) = 0
read(3, "A\"\0\0\0aeabi\0\1\30\0\0\0\0056\0\6\6\10\1\t\1\n\2\30\1\31\1\33"..., 35) = 35
_llseek(3, 9161672, [9161672], SEEK_SET) = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1160) = 1160
_llseek(3, 9161364, [9161364], SEEK_SET) = 0
read(3, "A\"\0\0\0aeabi\0\1\30\0\0\0\0056\0\6\6\10\1\t\1\n\2\30\1\31\1\33"..., 35) = 35
<CUT>
```

Investigation:

> _llseek(3, **9103084**, [9103084], SEEK_SET) = 0

9103084 is section header.

```

**NOT WORK SHARED OBJECT**
LC_ALL=C readelf -S ./libstable.so 
There are 29 section headers, starting at offset 0x8ae6ec:

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .ARM.exidx        ARM_EXIDX       00000194 000194 000010 00  AL 14   0  4
  [ 2] .dynsym           DYNSYM          000001a4 0001a4 0024d0 10   A  7   1  4
  [ 3] .gnu.version      VERSYM          00002674 002674 00049a 02   A  2   0  2
  [ 4] .gnu.version_r    VERNEED         00002b10 002b10 000140 00   A  7   6  4
  [ 5] .gnu.hash         GNU_HASH        00002c50 002c50 00074c 00   A  2   0  4
  [ 6] .hash             HASH            0000339c 00339c 001270 04   A  2   0  4
  [ 7] .dynstr           STRTAB          0000460c 00460c 0021d0 00   A  0   0  1
  [ 8] .rel.dyn          REL             000067dc 0067dc 03de20 08   A  2   0  4
  [ 9] .rel.plt          REL             000445fc 0445fc 000a38 08   A  2  20  4
  [10] .note.gnu.build-i NOTE            00045034 045034 000024 00   A  0   0  4
  [11] .rodata           PROGBITS        00045060 045060 0f7d60 00 AMS  0   0 32
  [12] .eh_frame_hdr     PROGBITS        0013cdc0 13cdc0 00000c 00   A  0   0  4
  [13] .eh_frame         PROGBITS        0013cdcc 13cdcc 000004 00   A  0   0  4
  [14] .text             PROGBITS        0013d000 13d000 743118 00  AX  0   0 64
  [15] .init             PROGBITS        00880118 880118 00000c 00  AX  0   0  4
  [16] .fini             PROGBITS        00880124 880124 000008 00  AX  0   0  4
  [17] .plt              PROGBITS        00880130 880130 001490 00  AX  0   0 16
  [18] .data             PROGBITS        00882000 882000 001454 00  WA  0   0  8
  [19] .tm_clone_table   PROGBITS        00883454 883454 000000 00  WA  0   0  4
  [20] .got.plt          PROGBITS        00883454 883454 000528 00  WA  0   0  4
  [21] .tbss             NOBITS          0088397c 88397c 00000c 00 WAT  0   0  4
  [22] .fini_array       FINI_ARRAY      00884000 884000 000004 04  WA  0   0  4
[... 56 lines cut ...]
difference in one section
 

> [27] .ARM.attributes   ARM_ATTRIBUTES  00000000 8bca94 000023 00      0   0  1

Okk, try run 
readelf -A ./libstable.so
`<empty result>`

readelf -A ./libbeta.so 
```
Attribute Section: aeabi
File Attributes
  Tag_CPU_name: "6"
  Tag_CPU_arch: v6
  Tag_ARM_ISA_use: Yes
  Tag_THUMB_ISA_use: Thumb-1
  Tag_FP_arch: VFPv2
  Tag_ABI_align_needed: 8-byte
  Tag_ABI_align_preserved: 8-byte, except leaf SP
  Tag_ABI_HardFP_use: Deprecated
  Tag_ABI_VFP_args: VFP registers
```

I think ld or dlopen check ".ARM.attributes" section and it fail when section is miss.

jwinarske, 2020-07-30:
I like your perseverance :1st_place_medal: 

What's different between the two toolchains are the pre-built libs.  The so link command is identical between the two.

I'm considering adding building Clang built-ins, and compiler-rt before the engine is linked.  This would minimize toolchain pre-built and target dependency to just crt1/crti/crtn.  I believe this will resolve the problem, as well as provide an optimized rt.

I'm out on vacation till mid next week, then I start a new job working with Flutter platform architecture!  So the soonest I will start this is next weekend.  I have the Clang build stages done in another project for Clang/MUSL, so it shouldn't be too bad.

DisDis, 2020-07-30:
It is very cool.  this also unlocks AOT for emb flutter, because snapshot_gen creates a "bad" libapp.so.

jwinarske, 2020-07-30:
Not sure what channel is not working, but I have AOT working in my Wayland flutter project.  I have steps documented to generate an AOT here:
https://github.com/jwinarske/meta-flutter/issues/7

DisDis, 2020-07-30:
Hmm, i used gen_snapshot from channel=dev and it generated "bad" shared object. Ok i look at Wayland project.

jwinarske, 2020-08-02:
@DisDis IIRC one of the host side tools does not build correctly.  I pulled the "bad" host-side tool from a manual x86_64 engine build.

jwinarske, 2021-09-11:
Closing this out due to inactivity

closed 2021-09-11 by jwinarske