meta-flutter/flutter_embedded #39: Ubuntu 20.04: "Performing download step for 'engine'" - error

DisDis, 2020-07-26:
Because engine.cmake contains
```
ExternalProject_Add(engine
    DOWNLOAD_COMMAND
        ${CMAKE_COMMAND} -E make_directory ${ENGINE_SRC_PATH} && cd ${ENGINE_SRC_PATH} &&
	PATH=${THIRD_PARTY_DIR}/depot_tools:$ENV{PATH} gclient config --spec ${GCLIENT_CONFIG} 
```
**PATH** is converted to '**"PATH**'
And we receive incorrect result:

> cd /home/dis/workspace/flutter/flutter_embedded/build/engine-prefix/src && /usr/bin/cmake -E make_directory /home/dis/workspace/flutter/flutter_embedded/third_party/engine && cd /home/dis/workspace/flutter/flutter_embedded/third_party/engine && **"PATH**=/home/dis/workspace/flutter/flutter_embedded/third_party/depot_tools:/home/dis/.local/bin:/home/dis/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/dis/bin/flutter/bin:/home/dis/Android/Sdk/platform-tools:~/.pub-cache/bin:/usr/lib/dart/bin:/home/dis/workspace/dart/sdk/depot_tools:/home/dis/workspace/dart/sdk/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin:/home/dis/.local/bin:/home/dis/bin/flutter/bin:/home/dis/Android/Sdk/platform-tools:~/.pub-cache/bin:/usr/lib/dart/bin:/home/dis/workspace/dart/sdk/depot_tools:/home/dis/workspace/dart/sdk/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin:/home/dis/.local/bin" gclient config --spec solutions=[{\"managed\":False,\"name\":\"src/flutter\",\"url\":\"git@github.com:flutter/engine.git\",\"custom_vars\":{\"download_android_deps\":False,\"download_windows_deps\":False,}}]

It work when i change engine.cmake:
```
ExternalProject_Add(engine
    DOWNLOAD_COMMAND
        ${CMAKE_COMMAND} -E make_directory ${ENGINE_SRC_PATH} && cd ${ENGINE_SRC_PATH} &&
	export PATH=${THIRD_PARTY_DIR}/depot_tools:$ENV{PATH} && gclient config --spec ${GCLIENT_CONFIG} &&
	export PATH=${THIRD_PARTY_DIR}/depot_tools:$ENV{PATH} && gclient sync --no-history --revision ${FLUTTER_ENGINE_SHA} -R -D -j ${NUM_PROC} -v
```

jwinarske, 2020-07-26:
Did you try using quotes like this?

 `PATH="${THIRD_PARTY_DIR}/depot_tools":"$ENV{PATH}"`

DisDis, 2020-07-27:
@jwinarske 
it is not work.
```
cd /home/dis/workspace/flutter/flutter_embedded/build/engine-prefix/src && /usr/bin/cmake -E make_directory /home/dis/workspace/flutter/flutter_embedded/third_party/engine && cd /home/dis/workspace/flutter/flutter_embedded/third_party/engine && "PATH=\"/home/dis/workspace/flutter/flutter_embedded/third_party/depot_tools\":\"/home/dis/.local/bin:/home/dis/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/dis/bin/flutter/bin:/home/dis/Android/Sdk/platform-tools:~/.pub-cache/bin:/usr/lib/dart/bin:/home/dis/workspace/dart/sdk/depot_tools:/home/dis/workspace/dart/sdk/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin:/home/dis/.local/bin:/home/dis/.local/bin\"" gclient config --spec solutions=[{\"managed\":False,\"name\":\"src/flutter\",\"url\":\"git@github.com:flutter/engine.git\",\"custom_vars\":{\"download_android_deps\":False,\"download_windows_deps\":False,}}] && "PATH=\"/home/dis/workspace/flutter/flutter_embedded/third_party/depot_tools\":\"/home/dis/.local/bin:/home/dis/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/dis/bin/flutter/bin:/home/dis/Android/Sdk/platform-tools:~/.pub-cache/bin:/usr/lib/dart/bin:/home/dis/workspace/dart/sdk/depot_tools:/home/dis/workspace/dart/sdk/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin:/home/dis/.local/bin:/home/dis/.local/bin\"" gclient sync --no-history --revision ee76268252c22f5c11e82a7b87423ca3982e51a7 -R -D -j 32 -v
/bin/sh: 1: PATH="/home/dis/workspace/flutter/flutter_embedded/third_party/depot_tools":"/home/dis/.local/bin:/home/dis/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/dis/bin/flutter/bin:/home/dis/Android/Sdk/platform-tools:~/.pub-cache/bin:/usr/lib/dart/bin:/home/dis/workspace/dart/sdk/depot_tools:/home/dis/workspace/dart/sdk/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin:/home/dis/.local/bin:/home/dis/.local/bin": not found

```

jwinarske, 2020-07-27:
echo $SHELL

DisDis, 2020-07-27:
echo $SHELL
`/bin/bash`
uname -a
`Linux dark-star 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`
Ubuntu 20.04 64Bit

@jwinarske

jwinarske, 2020-07-27:
`git pull` on master.  You should be good to go.

DisDis, 2020-07-27:
@jwinarske i get other issue

> make package -j8

```
[ 16%] Built target sysroot
[ 32%] Built target depot_tools
[ 48%] Built target symlink_fixups
[ 51%] Performing build step for 'engine'
ninja: Entering directory `out/linux_release_arm'
[4/806] AR obj/third_party/dart/runtime/libdart_precompiled_runtime.a
FAILED: obj/third_party/dart/runtime/libdart_precompiled_runtime.a 
rm -f obj/third_party/dart/runtime/libdart_precompiled_runtime.a && /home/dis/workspace/flutter/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/armv7-neon-vfpv4-linux-gnueabihf-ar rcs obj/third_party/dart/runtime/libdart_precompiled_runtime.a @obj/third_party/dart/runtime/libdart_precompiled_runtime.a.rsp
/bin/sh: 1: /home/dis/workspace/flutter/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/armv7-neon-vfpv4-linux-gnueabihf-ar: not found
[5/806] AR obj/third_party/libjpeg-turbo/libsimd.a
FAILED: obj/third_party/libjpeg-turbo/libsimd.a 
rm -f obj/third_party/libjpeg-turbo/libsimd.a && /home/dis/workspace/flutter/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/armv7-neon-vfpv4-linux-gnueabihf-ar rcs obj/third_party/libjpeg-turbo/libsimd.a @obj/third_party/libjpeg-turbo/libsimd.a.rsp
/bin/sh: 1: /home/dis/workspace/flutter/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang/bin/armv7-neon-vfpv4-linux-gnueabihf-ar: not found
[35/806] CC obj/third_party/libxml/src/libxml2.parser.o
ninja: build stopped: subcommand failed.
make[2]: *** [CMakeFiles/engine.dir/build.make:112: engine-prefix/src/engine-stamp/engine-build] Ошибка 1
make[1]: *** [CMakeFiles/Makefile2:168: CMakeFiles/engine.dir/all] Ошибка 2
make: *** [Makefile:152: all] Ошибка 2
```

jwinarske, 2020-07-28:
@DisDis those are not related.  See https://github.com/jwinarske/flutter_embedded/issues/38

closed 2020-07-28 by jwinarske

DisDis, 2020-07-28:
but MR contain mistacke, because you miss 

> PATH=${THIRD_PARTY_DIR}/depot_tools:$ENV{PATH} 

, after i changed code, the engine is compiled.
@jwinarske

jwinarske, 2020-07-28:
See the changes branch.  After a few more tests it will merge into master.