meta-flutter/flutter_embedded #41: Step 'flutter-pi' doesn't work

DisDis, 2020-07-27:
cmake .. -DCHANNEL=stable -DENGINE_RUNTIME_MODE=debug -DBUILD_FLUTTER_PI=ON -DBUILD_GLFW_FLUTTER=OFF

```
-- CMAKE_BUILD_TYPE not set, defaulting to Release.
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Generator .............. Unix Makefiles
-- Build Type ............. Release
-- TARGET_ARCH not set, defaulting to arm
-- TARGET_SYSROOT not set, defaulting to /home/dis/workspace/flutter/flutter_embedded/sysroot
-- Rootfs Archive Url ..... http://director.downloads.raspberrypi.org/raspbian/archive/2020-02-14-13:48/root.tar.xz
-- Target Sysroot ......... /home/dis/workspace/flutter/flutter_embedded/sysroot
-- Engine Repo ............ git@github.com:flutter/engine.git
-- Flutter Channel ........ stable
-- Engine SHA1 ............ ee76268252c22f5c11e82a7b87423ca3982e51a7
-- ENGINE_LIB_FLAGS........ -Wl,-z,notext -nostdlib++ -fuse-ld=lld
-- PKG_CONFIG_PATH ........ /home/dis/workspace/flutter/flutter_embedded/sysroot/usr/lib/arm-linux-gnueabihf/pkgconfig:/usr/share/pkgconfig
-- Engine Flags ........... --runtime-mode debug --no-goma --lto --clang --enable-fontconfig --embedder-for-target --full-dart-sdk --target-os linux --linux-cpu arm --target-sysroot /home/dis/workspace/flutter/flutter_embedded/sysroot --target-toolchain /home/dis/workspace/flutter/flutter_embedded/third_party/engine/src/buildtools/linux-x64/clang --target-triple armv7-neon-vfpv4-linux-gnueabihf --arm-float-abi hard
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dis/workspace/flutter/flutter_embedded/build

```

Output:
```
[ 87%] No update step for 'flutter-pi'
[ 87%] Performing patch step for 'flutter-pi'
cd /home/dis/workspace/flutter/flutter_embedded/build/flutter-pi-prefix/src/flutter-pi && /usr/bin/cmake -E copy /home/dis/workspace/flutter/flutter_embedded/cmake/flutter.pi.cmake /home/dis/workspace/flutter/flutter_embedded/build/flutter-pi-prefix/src/flutter-pi/CMakeLists.txt && /usr/bin/cmake -E copy /home/dis/workspace/flutter/flutter_embedded/build/stable/flutter_embedder.h /home/dis/workspace/flutter/flutter_embedded/build/flutter-pi-prefix/src/flutter-pi-build
cd /home/dis/workspace/flutter/flutter_embedded/build/flutter-pi-prefix/src/flutter-pi && /usr/bin/cmake -E echo_append
cd /home/dis/workspace/flutter/flutter_embedded/build/flutter-pi-prefix/src/flutter-pi && /usr/bin/cmake -E touch /home/dis/workspace/flutter/flutter_embedded/build/flutter-pi-prefix/src/flutter-pi-stamp/flutter-pi-update
Error copying file "/home/dis/workspace/flutter/flutter_embedded/build/stable/flutter_embedder.h" to "/home/dis/workspace/flutter/flutter_embedded/build/flutter-pi-prefix/src/flutter-pi-build".
make[2]: *** [CMakeFiles/flutter-pi.dir/build.make:102: flutter-pi-prefix/src/flutter-pi-stamp/flutter-pi-patch] Ошибка 1
make[2]: *** Ожидание завершения заданий…
make[2]: выход из каталога «/home/dis/workspace/flutter/flutter_embedded/build»
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/flutter-pi.dir/all] Ошибка 2
make[1]: выход из каталога «/home/dis/workspace/flutter/flutter_embedded/build»
make: *** [Makefile:152: all] Ошибка 2

```

I tried to change flutter-pi.cmake
```
-        ${CMAKE_COMMAND} -E copy ${CMAKE_BINARY_DIR}/${CHANNEL}/${ENGINE_HEADER} ${FLUTTER_PI_SOURCE_DIR}-build
+        ${CMAKE_COMMAND} -E copy ${THIRD_PARTY_DIR}/engine/src/${ENGINE_OUT_DIR}/${ENGINE_HEADER} ${FLUTTER_PI_SOURCE_DIR}-build 
```
But it didn't change 'make' output.
How do I change the 'flutter-pi' step?

jwinarske, 2020-07-28:
@DisDis it won't build as you specify as there are dependent libs needed for flutter-pi that are not in the default Raspbian rootfs.
stable channel is default, as is debug for engine runtime.  It should look something like this:

`cmake .. -GNinja -DBUILD_FLUTTER_PI=ON -DTARGET_SYSROOT=/path/to/you/fixed/up/symlink/rootfs/with/required/packages -DBUILD_PLATFORM_SYSROOT=OFF`

Also was this on a clean build folder, or previously used?  If previously used, what was the prior build setup?

DisDis, 2020-07-28:
> Also was this on a clean build folder, or previously used? If previously used, what was the prior build setup?

I don't remembe, but i think it was around -DCHANNEL=stable -DENGINE_RUNTIME_MODE=debug -DBUILD_FLUTTER_PI=ON -DBUILD_GLFW_FLUTTER=OFF

How do i clean build? (cmake clean??, make clean???)

jwinarske, 2020-07-28:
cd build
rm -rf *

It doesn't blow away the engine repo, and that will sync automatically.

jwinarske, 2020-07-28:
https://github.com/jwinarske/flutter_embedded/pull/42

closed 2020-07-28 by jwinarske